MarlinFirmware/Marlin #11482: Z position unknown after autohome

ryandeleon1, 2018-08-07:
I recently updated the firmware on my Gmax Gcreate 1.5+ to Marlin 1.1.8

Now after autohoming the printer the x and y positions read the correct values but the z position will read a set of characters. "*,-(" on the LCD screen. After this happens the printer will not move in the z direction and autohoming the printer again causes head to raise in the z direction indefinitely until it is turned off.

thinkyhead, 2018-08-07:
### Z: *,-(

So, Z is obviously at Pirate height.

What does `M114` say after you do a `G28`?

thriftynuts, 2018-08-20:
Im having the same problem. M114 returns:

X:197.00 Y:228.00 Z:0.00 E:0.00
Count X:15760 Y:18240 Z:-214783648

thinkyhead, 2018-08-21:
Do you see the same issue in 1.1.9?

thinkyhead, 2018-09-09:
Try updating their configuration files for 1.1.9, which has hundreds of patches since 1.1.8. We don't patch old versions, so 1.1.8 will forever be as it is today.

boelle, 2019-02-20:
@ryandeleon1 is problem gone if you use bugfix 2.0?

boelle, 2019-03-13:
@thinkyhead think we can close this one

closed 2019-03-14 by thinkyhead