MarlinFirmware/Marlin #11483: [2.0.X] LPC1768 Serial port error

ModMike, 2018-08-07:
Trying to use serial port 1 for ESP32 Web UI by Luc to communicate with Marlin.  Web UI properly configured and Marlin selected as target firmware.

### Steps to Reproduce

#define SERIAL_PORT 1

```
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/LPC1768_PWM.o
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../pins/pins.h:926:0,

from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:30,
from Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp:51:
Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp: In function 'void spiBegin()':
Marlin/src/HAL/HAL_LPC1768/../../inc/../pins/../HAL/HAL_LPC1768/spi_pins.h:42:29: error: 'P0_15' was not declared in this scope
#define SCK_PIN           P0_15
^
Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp:136:37: note: in expansion of macro 'SCK_PIN'
PinCfg.Pinnum = LPC1768_PIN_PIN(SCK_PIN);
^~~~~~~
Marlin/src/HAL/HAL_LPC1768/../../inc/../pins/../HAL/HAL_LPC1768/spi_pins.h:42:29: note: suggested alternative: 'P0_11'
#define SCK_PIN           P0_15
^
Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp:136:37: note: in expansion of macro 'SCK_PIN'
PinCfg.Pinnum = LPC1768_PIN_PIN(SCK_PIN);
^~~~~~~
*** [.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_spi.o] Error 1

```

gloomyandy, 2018-08-07:
What board are you using? Looking at the serial configuration serial port 1 will use pin P0_15 and so it is not available as the default pin for spi use (see spi_pins.h). Are you using SPI access to anything (like an external SD card reader)? Can you use a different serial port to avoid the clash?

ModMike, 2018-08-07:
Sorry, I forgot to mention it’s an Aztec X5 Mini.  Not using SPI at all.

ModMike, 2018-08-07:
I've attached all the files Ive modified.  This will compile perfectly until you change the serial port.

SCK was undefined and even if I define it, it's stuck in an if loop that doesn't execute.  I tried moving out of the loop but it does not seem to make a difference.  Can you take a peek?

forkoz, 2018-08-07:
files?

ModMike, 2018-08-07:
Don't know if this sheds any light on the issue but the 8266 RX/TX are jumped to the MCU, not RX0 & TX0

ModMike, 2018-08-07:
Oops!  I attached them in the original post.

forkoz, 2018-08-07:
the wiring diagram doesn't tell you where those serial pins go or what serial port they use.

in pinmapping.h
#if SERIAL_PORT != 1 && SERIAL_PORT_2 != 1
 #define P0_15 LPC1768_PIN(PORT(0), PIN(15), INTERRUPT(1), PWM(0), ADC_NONE)
 #define P0_16 LPC1768_PIN(PORT(0), PIN(16), INTERRUPT(1), PWM(0), ADC_NONE)

which is the same as your SPI pins in your pin file and the default pins (spi_pins.h):
    //#define MISO_PIN            P0_17   // (50)  system defined J3-10 & AUX-3
    //#define MOSI_PIN            P0_18   // (51)  system defined J3-10 & AUX-3
    //#define SCK_PIN             P0_15   // (52)  system defined J3-9 & AUX-3
    //#define SS_PIN              P1_23   // (53)  system defined J3-5 & AUX-3 - sometimes called SDSS

so you have to do something about p0_15 or not use serial port 1. They can't be used for 2 things at the same time.

From looking at the schematic of the board, I think they are using serial port 3 just like sbase.
http://files.panucatt.com/datasheets/x5mini_v1.0_designfiles.zip

ModMike, 2018-08-07:
Those are commented out so should be no conflict.  So I need to define serial 3?

forkoz, 2018-08-07:
they're commented out in your pins file but not in spi_pins.h. yea, get rid of serial 1 and use serial3, they go to specific pins anyway. USB is -1, its all in pinmapping.h. even if it were to compile your serial port isn't hooked up there unless you're connecting the esp to the EXP2 connector.

ModMike, 2018-08-07:
Whats the best way of finding out those pin names?  Any guesses?

forkoz, 2018-08-07:
pinmapping.h in include folder tells you what the pins per port are. then you can look at the schematic.
 that's how I deduced it being serial 3 and not serial 1. If you are using EXP2 then you'll have to define a different  SCK_PIN so it overrides the default.

ModMike, 2018-08-07:
I appreciate your help and patience with a non-programmer such as myself.  I will try to soldier on!

ModMike, 2018-08-07:
Is there anyway to deduce this from smoothieware?

forkoz, 2018-08-07:
probably, the config file makes you put all the pin numbers in for everything.

ModMike, 2018-08-07:
I was browsing through it and nothing for serial is there.  This is the sample file for Mini.

[config.txt](https://github.com/MarlinFirmware/Marlin/files/2268401/config.txt)

ModMike, 2018-08-07:
Looks like P0.18 on EXP2 would do it, but also used for display.

forkoz, 2018-08-08:
what happens when you enable serial3 and hook up the pins?

timeshell, 2018-08-08:
I'm not sure if this is applicable, but installing Marlin 2 on the Re-ARM, using the LPC176x processor, requires setting SERIAL to -1 for the built in USB.  To use the pins serial, I defined SERIAL1 as 0.  This worked for me on the Smoothieboard.

timeshell, 2018-08-08:
Also take a look at the last post here:

https://reprap.org/forum/read.php?13,603758

ModMike, 2018-08-08:
DUDE!!!!  It Works!  I know have a board that runs marlin with a built in WEB UI, how awesome is that!

#define SERIAL_PORT -1
#define SERIAL_PORT_2 0

Thank you Timeshell!

A few tips for other trying this.  

1)  I am currently testing 250000 baud rate.  At first it seems like it does not work but reset and give it some time to boot.  It's usually ok when you see the SD card on your computer if connected by USB.

2) Flashing Marlin on this board does not seem to be as simple as hitting reset.  It appears its a dual reset process.  First save the firmware to SD using PIO Upload, then use reset.  You should see the LEDs flash and go red.  Now reset again and they should all be green.

I will be amending my original guide with a chapter on Azteeg WBE UI.

Now I need to see if I can get the TMC2224 (identical to TMC2208 but different pin configuration) to talk to Marlin.

closed 2018-08-08 by ModMike