MarlinFirmware/Marlin #11484: [1.1.9] Is there a problem with 2 mixing extruders ? (M165 use)

Balzac40, 2018-08-07:
### Description

On printing, extruder0 runs continuously and at an anormal high speed… Seems to ignore retractions also.
Trying to modify this speed by issuing a M165 command like "M165 A0.1 B0.9", E0 speed increases more (should decrease) while E1 starts like E0 at the beginning…

### Steps to Reproduce

Printer : CTC Prusa I3 Pro B, board GT2560 Rev A+, A4988 drivers, Cyclops Hot end

Marlin 1.1.9 : Activate mixing extruders
ALL OTHERS NEW FUNCTIONS NOT USED !

Reverting back to Marlin 1.1.8 stops this problem.

sergiofagundes, 2018-08-07:
I'm also having this proble. I have a Delta with Rumba board, TMC2100 drivers, Diamond hotend 3in1out.

The steppers moves too fast.

thinkyhead, 2018-08-13:
Can you try bugfix-2.0.x and see if the issue exists in that version?

Balzac40, 2018-08-13:
Thanks thinkyhead for your answer!
But my board has a 8bits processor... Not a problem with bugfix-2.0.x ?

AnHardt, 2018-08-13:
`bugfix-2.0.x` is 8 and 32bit!

Balzac40, 2018-08-13:
Sorry for this mistake !
Will make a try then ASAP

Thanks !

Balzac40, 2018-08-18:
Was unable to use bugfix-2.0.x, sorry !
With Atom problem with tk/tcl during PIO Build… Did'nt find a solution or workaround…
With Arduino IDE, 3 different (W10 app, 1.8.5, 1.9.0) versions tested, no more success !
Staying back to 1.1.8...

Balzac40, 2018-08-19:
Finally, got bugfix-2.0.x on my CTC printer.
And I confirm : the problem is the same with this release.
The issue is the same. Seems to me that extruders run faster than with 1.1.9.

Balzac40, 2018-09-29:
On september 28, I made a new try with bugfix-2.0.x on my CTC prusa printer.
No change. Issue is the same. (2 extruders, one hotend mixing colors)
Prior to launch print, I type M165 A0.5 B0.5 (Repetier Host)
When printing starts, extruders steppers motors runs continuously… with an very excessive speed of course…

Files : configuration.h and adv.h
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2430298/Marlin.zip)

thinkyhead, 2018-09-29:
Can I borrow your printer? I'm shooting in the dark here.

Balzac40, 2018-09-29:
Sending my printer from France would be… difficult!
It's a first price printer (kit to assemble). Board GT2560 Rev A+. Drivers A4988. Nema 17 motors.
Powered 12V for board, fans and motors, and 24V for hotend and hotbed thru external Mosfets…
I use also TL-smoothers+ from TriangleLab… on all motors.
With Marlin 1.1.8... all is ok

thinkyhead, 2018-09-29:
Yeah, the mixing extruder stuff got a little beat up in the 1.1.9 planner/stepper refactor. We've been trying to get it patched up, and we thought we got it in the last couple of days. I'm saddened to hear it still isn't working as it should. It's tricky to debug without the requisite hardware, but we'll keep trying.

Balzac40, 2018-09-29:
Sorry. I would like to help more… But I don't use C since… uh uh… 30 years.

AnHardt, 2018-09-29:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/planner.cpp#L1771-L1780

Try 
```
      block->mix_steps[i] = mixing_factor[i] * esteps;

```
instead.

AnHardt, 2018-09-29:
I'm working on a new MixingExtruder implementation, but will last, at least, another week.

The above is also the minimum needed improvement in the V1 branch.

thinkyhead, 2018-09-29:
Thanks, Andreas. I’ll apply the patch ASAP.

thinkyhead, 2018-10-06:
Ok, after a number of patches, maybe mixing is back in business? Please check the `bugfix-2.0.x` branch and see how we're doing.

Balzac40, 2018-10-07:
Want to check again but I encounter an error I don’t understand… (PatformIo Build)

« « « 
In file included from Marlin\src\module\../lcd/ultralcd.h:42:0,
from Marlin\src\module\probe.cpp:39:
Marlin\src\module\probe.cpp: In function 'void probing_pause(bool)':
Marlin\src\module\../lcd/../Marlin.h:50:54: error: 'X_ENABLE_WRITE' was not declared in this scope
#define disable_X() do{ X_ENABLE_WRITE(!X_ENABLE_ON); CBI(axis_known_position, X_AXIS); }while(0)
^
Marlin\src\module\probe.cpp:298:9: note: in expansion of macro 'disable_X'
disable_X(); disable_Y();
^
Marlin\src\module\../lcd/../Marlin.h:61:54: error: 'Y_ENABLE_WRITE' was not declared in this scope
#define disable_Y() do{ Y_ENABLE_WRITE(!Y_ENABLE_ON); CBI(axis_known_position, Y_AXIS); }while(0)
^
Marlin\src\module\probe.cpp:298:22: note: in expansion of macro 'disable_Y'
disable_X(); disable_Y();

« « «

Balzac40, 2018-10-13:
Hurrah ! It's ok ! Extruders run correctly and respond to M165 orders!
Nice job! Thank you very much

closed 2018-10-15 by Balzac40

orhaan68, 2018-11-12:
Dear friends, i just flash my 3D printer (ramps 1.4) with marlin bugfix 2.0.x . Previously i was using marlin 1.1.9 and i have problems with diamond hotand and mixing extruder. At 1.1.9 I used mixing ratio at repetier host with M163 and M164 commands. But steppers moves more than command at mixing. At bugfix 2.0.x steppers move as command at mixing just with M165 command. M163 and M164 do not works. When I select virtual extruder T0 or T1 etc, it said unknown command. Any idea?
Thank you.

thinkyhead, 2018-11-12:
Why are you posting on a closed issue? You should open a new issue and fill out the template so that we have all the information we need to proceed with an investigation. Please follow that procedure.

AnHardt, 2018-11-12:
M164 syntax is
```
M164 S0
```
or 
```
M164 S1
```
...
```
M164 ; without parameter to save to the tool currently in use.
```
not
```
M164 T1
```
or
```
M164 ST1
```

orhaan68, 2018-11-12:
Sorry about posting on a closed issue. I am pretty new at "github.com". I move issue at following link.
https://github.com/MarlinFirmware/Marlin/issues/12409

commit 14df7530 (2018-09-30): Correction for block->mix_steps

Fix #11484
commit 0ad8646a (2018-09-30): Fix block->mix_steps

Fix #11484