MarlinFirmware/Marlin #11487: 1.1.9 doesn't seem to play nice with corexy

rjcuomo1, 2018-08-07:
### Description

I uploaded the latest version of Marlin and all the movements are messed up. It's almost like the printer fights itself. I tried both 1.1.9 and 1.1.9 bugfix. I haven't changed any settings I had in 1.1.8 except turning on S Curve and inputting driver type (DRV8825). I even turned off S Curve just to verify it wasn't the problem.

### Steps to Reproduce

1. Upgrade to 1.1.9 from 1.1.8
2. Upgrade to 1.1.9 bugfix from 1.1.9
3. Go back to 1.1.8 because it works right

Bed leveling with 1.1.8
**Expected behavior:** https://youtu.be/9Wf_oSivrXo

Bed leveling with 1.1.9
**Actual behavior:** https://youtu.be/nyQylGtb7ZU

If I just let it run, it layer shifts every couple layers

#### Additional Information

timeshell, 2018-08-08:
Reload defaults and re-save to EEPROM?

rjcuomo1, 2018-08-15:
I'm not sure how that will help when uploading a new version changes the defaults

thinkyhead, 2018-08-21:
If you did `M502`, `M500` and it didn't help, then give `bugfix-2.0.x` a try.

cfelicio, 2018-08-22:
Umm, its working fine for me (I have a D Bot), can you share your configuration files?

The only issue I have so far is with sudden layer shifting on long prints after a few hours, but still not sure what's causing it (trying to check all mechanical and electrical stuff first). I suspect a driver must be overheating at this point.

rjcuomo1, 2018-09-05:
Does bugfix-2.0.x not have all the files that the 1.1.x downloads have? I opened it up to only see the configuration.h and configuration_adv.h files. There aren't any of the other .h or .cpp files.

thinkyhead, 2018-09-06:
> Does bugfix-2.0.x not have all the files that the 1.1.x downloads have? I opened it up to only see the configuration.h and configuration_adv.h files. There aren't any of the other .h or .cpp files.

They have been moved to reduce clutter in the Arduino IDE. You can view and edit the rest of the files in another text editor or IDE, such as Notepad++, Sublime Text 3, Atom, or Visual Studio Code.

boelle, 2019-02-20:
@rjcuomo1 is problem gone in bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead