MarlinFirmware/Marlin #11489: [2.0.x]  Configure ESP8266 serial port

ModMike, 2018-08-08:
Hardware: Azteeg X5 Mini Wifi. 

Environment: 

1) ESP3D-WEBUI is running on 8266 , accessible from browser, and configured for Marlin.   
2) Marlin is running and accessible from USB

Issue: WEBUI is unable to talk to Marlin due to serial config.  Attempting set serial_port 1 fails to compile:

Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/LPC1768_PWM.o
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../pins/pins.h:926:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:30,
from Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp:51:
Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp: In function 'void spiBegin()':

Marlin/src/HAL/HAL_LPC1768/../../inc/../pins/../HAL/HAL_LPC1768/spi_pins.h:42:29: error: 'P0_15' was not declared in this scope
#define SCK_PIN           P0_15
^
Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp:136:37: note: in expansion of macro 'SCK_PIN'
PinCfg.Pinnum = LPC1768_PIN_PIN(SCK_PIN);
^~~~~~~
Marlin/src/HAL/HAL_LPC1768/../../inc/../pins/../HAL/HAL_LPC1768/spi_pins.h:42:29: note: suggested alternative: 'P0_11'
#define SCK_PIN           P0_15
^
Marlin/src/HAL/HAL_LPC1768/HAL_spi.cpp:136:37: note: in expansion of macro 'SCK_PIN'
PinCfg.Pinnum = LPC1768_PIN_PIN(SCK_PIN);
^~~~~~~
*** [.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/HAL_spi.o] Error 1

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2268499/Archive.zip)

closed 2018-08-08 by ModMike

ModMike, 2018-08-08:
Thank you to Timeshell for the help.

#define SERIAL_PORT -1
#define SERIAL_PORT_2 0

A few tips for others trying this:

I am currently testing 250000 baud rate. At first it seems like it does not work but reset and give it some time to boot. It's usually ok when you see the SD card on your computer if connected by USB.

Flashing Marlin on this board does not seem to be as simple as hitting reset. It appears its a dual reset process. First save the firmware to SD using PIO Upload, then use reset. You should see the LEDs flash and go red. Now reset again and they should all be green.

Unfortunately reset does not always seem to work. A sure fire method is to cut power.

I will be amending my original guide with a chapter on Azteeg WEB UI.

timeshell, 2018-08-08:
Your welcome ModMike.

Reference to #11483