MarlinFirmware/Marlin #11490: [bugfix-2.0.x] Temperature oscillates if set close to 255, PWM behaves like bang-bang

VanessaE, 2018-08-08:
At commit 902167fa

### Description

If the hotend temperature is set to 255° (or perhaps within a couple of degrees of that), the PID/PWM code kinda chokes.  That is, the hotend will heat up just fine, and it can go much hotter if I want (though this is already close to the hotend's safe limit), but the temperature never stabilizes once it reaches the target.

The hotend is properly PID-tuned, though I re-ran the tuning, just to make sure that wasn't the cause.

It does try to settle as it approaches the target temperature, but it starts to oscillate immediately, growing in amplitude to a maximum of about ±6° or so after about 2 minutes.  By that point, PWM values are mostly only 0 or 127, so it's basically behaving like bang-bang mode.

The printer can print just fine in this situation, though such wide temperature variations manifest as variations in shininess and quality in the print.  Getting the print to start sometimes takes a while, due to the 10-second "wait for temperature to settle" timer, which gets reset repeatedly if the temperature is oscillating too much at the start (I'm not sure if Pronterface imposes this timer, or if it's a firmware thing).

If I drop the temperature setting to 250°, the oscillations start to diminish, stabilizing adequately by about 3 minutes out.  After another minute or two, it's dead-on.

Since this happens at 255°, I can only assume it's a simple 8-bit overflow error somewhere.

### Steps to Reproduce

Just set the hotend to 255°, and observe the temperature readings, PWM values, and graph in e.g. Pronterface.

**Expected behavior:**

It should rise to 255 and settle there after a moment.

**Actual behavior:**

It swings back and forth like a trapeze. :smiley:

eamaclean, 2018-10-02:
A friend of mine just ran into exactly this issue in the bugfix-1.1.x branch on a modified CR-10. Interestingly, entering the menu on the LCD apparently kicks it out of this state.

It's led to his printer being sat hot and not printing for 2-3 hours at least once already.

gloomyandy, 2018-10-02:
What target temperature did you use when tuning your PID values? Have you tried doing the tuning with 255 as the target temperature? What PID values did it come up with?

VanessaE, 2018-10-02:
Normally I tune with the target set somewhere close to 230.

VanessaE, 2018-10-02:
```c
  #define DEFAULT_Kp 16.2
  #define DEFAULT_Ki 1.2
  #define DEFAULT_Kd 60.1
```

VanessaE, 2018-10-02:
Given the swinging, I did not dare try anything higher than that.

VanessaE, 2018-12-07:
Any movement/updates on this?

(@thinkyhead poking you since you're active :smiley: )

VanessaE, 2018-12-07:
Actually, strike that.  I've updated (to commit 055cb2b9563d02a6df709a803026242dfc1ea603) and configured, and this no longer appears to be a problem.

closed 2018-12-07 by VanessaE