MarlinFirmware/Marlin #11491: Unable To Compile after adding LCD Marlin-bugfix-2.0.x

rkozak, 2018-08-08:
Attempting to compile Marlin-bugfix-2.0.x for TM2130 on X & Y. Compiled successfully until I enables the LCD (#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER).

Compile error received....

Arduino: 1.8.1 (Windows 7), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Build options changed, rebuilding all
sketch\src\lcd\lcdprint_u8g.cpp: In function 'int lcd_put_wchar_max(uint32_t, pixel_len_t)':

sketch\src\lcd\lcdprint_u8g.cpp:41:26: error: 'class U8GLIB' has no member named 'getPrintCol'

   unsigned int x = pu8g->getPrintCol(),

                          ^

sketch\src\lcd\lcdprint_u8g.cpp:44:25: error: 'ret' was not declared in this scope

   pu8g->setPrintPos(x + ret, y);

                         ^

sketch\src\lcd\lcdprint_u8g.cpp:44:30: error: 'y' was not declared in this scope

   pu8g->setPrintPos(x + ret, y);

                              ^

sketch\src\lcd\lcdprint_u8g.cpp: In function 'int lcd_put_u8str_max(const char*, pixel_len_t)':

sketch\src\lcd\lcdprint_u8g.cpp:50:26: error: 'class U8GLIB' has no member named 'getPrintCol'

   unsigned int x = pu8g->getPrintCol(),

                          ^

sketch\src\lcd\lcdprint_u8g.cpp:53:25: error: 'ret' was not declared in this scope

   pu8g->setPrintPos(x + ret, y);

                         ^

sketch\src\lcd\lcdprint_u8g.cpp:53:30: error: 'y' was not declared in this scope

   pu8g->setPrintPos(x + ret, y);

                              ^

sketch\src\lcd\lcdprint_u8g.cpp: In function 'int lcd_put_u8str_max_P(const char*, pixel_len_t)':

sketch\src\lcd\lcdprint_u8g.cpp:58:26: error: 'class U8GLIB' has no member named 'getPrintCol'

   unsigned int x = pu8g->getPrintCol(),

                          ^

sketch\src\lcd\lcdprint_u8g.cpp:61:25: error: 'ret' was not declared in this scope

   pu8g->setPrintPos(x + ret, y);

                         ^

sketch\src\lcd\lcdprint_u8g.cpp:61:30: error: 'y' was not declared in this scope

   pu8g->setPrintPos(x + ret, y);

                              ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

rkozak, 2018-08-22:
Can someone help with this?

yhfudev, 2018-09-10:
From the error messages, it seems that your U8GLIB is not correct version.

I compiled the latest 2.x branch (Sep 10, 7391e233324baea43597345cf72b91387762f707) successfully with the default LCD (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER).

croaker-1, 2018-12-15:
I just ran into the same error using U8glib 1.19.1 (the latest in the Arduino library manager). I downgraded to 1.18.0 and it complied without any problems.

boelle, 2019-02-20:
@rkozak is this still an issue?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead