MarlinFirmware/Marlin #11493: [bugfix2.0.x] TMC2208 compile error

ModMike, 2018-08-08:
### Environment

Azteeg X5 Mini WiFi
Marlin bug fix 2.0.x version - Aug 8th, 2018

### Description

#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208

etc...

produces following error:

```
Compiling .pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/u8g/HAL_LCD_I2C_routines.o
Marlin/src/HAL/HAL_LPC1768/main.cpp: In function 'int main()':
Marlin/src/HAL/HAL_LPC1768/main.cpp:104:29: error: no matching function for call to 'HalSerial::begin()'
MYSERIAL0.begin(BAUDRATE);
^
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_LPC1768/HAL.h:67:0,

from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/main.cpp:27:
Marlin/src/HAL/HAL_LPC1768/include/serial.h:105:8: note: candidate: void HalSerial::begin(int32_t)
void begin(int32_t baud) { }
^~~~~
Marlin/src/HAL/HAL_LPC1768/include/serial.h:105:8: note:   candidate expects 1 argument, 0 provided
Marlin/src/HAL/HAL_LPC1768/main.cpp:106:31: error: no matching function for call to 'HardwareSerial::begin()'
MYSERIAL1.begin(BAUDRATE);
^
In file included from Marlin/src/HAL/HAL_LPC1768/../../inc/../HAL/HAL_LPC1768/HAL.h:69:0,
from Marlin/src/HAL/HAL_LPC1768/../../inc/MarlinConfig.h:28,
from Marlin/src/HAL/HAL_LPC1768/main.cpp:27:
Marlin/src/HAL/HAL_LPC1768/include/HardwareSerial.h:74:8: note: candidate: void HardwareSerial::begin(uint32_t)
void begin(uint32_t baudrate);
^~~~~
Marlin/src/HAL/HAL_LPC1768/include/HardwareSerial.h:74:8: note:   candidate expects 1 argument, 0 provided
*** [.pioenvs/LPC1768/src/src/HAL/HAL_LPC1768/main.o] Error 1
```
[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2270155/Archive.zip)

teemuatlut, 2018-08-08:
You've removed the defined value from `BAUDRATE`.

ModMike, 2018-08-08:
I have no idea how that happened, but fixed and back to old issue:

Compiling .pioenvs/LPC1768/src/src/module/tool_change.o
Marlin/src/module/stepper_indirection.cpp:303:32: error: 'X_SERIAL_RX_PIN' was not declared in this scope
_TMC2208_DEFINE_SOFTWARE(X);
^
Marlin/src/module/stepper_indirection.cpp:295:93: note: in definition of macro '_TMC2208_DEFINE_SOFTWARE'
#define _TMC2208_DEFINE_SOFTWARE(ST) SoftwareSerial ST##_HARDWARE_SERIAL = SoftwareSerial(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN); \
^~
Marlin/src/module/stepper_indirection.cpp:303:32: note: suggested alternative: '__SERIAL_H__'
_TMC2208_DEFINE_SOFTWARE(X);
^
Marlin/src/module/stepper_indirection.cpp:295:93: note: in definition of macro '_TMC2208_DEFINE_SOFTWARE'
#define _TMC2208_DEFINE_SOFTWARE(ST) SoftwareSerial ST##_HARDWARE_SERIAL = SoftwareSerial(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN); \
^~
Marlin/src/module/stepper_indirection.cpp:303:32: error: 'X_SERIAL_TX_PIN' was not declared in this scope

ModMike, 2018-08-08:
Oops!!!!!  I realized and fixed it but posted wrong error.  I updated original post to reflect issue.

teemuatlut, 2018-08-08:
> 'Y_SERIAL_RX_PIN' was not declared

You haven't defined the SW UART pins that you'd like to use with the driver.

ModMike, 2018-08-08:
That definitely fixes the compile error but broke the WEB UI because of incorrect pin assignments on my part.  I will get the info from Panucatt and keep everyone posted.

Thank you for the quick replies!

ModMike, 2018-08-08:
So I have been at this all day and am at a loss how to proceed.  I need to define the following correct?  Sample form MKS Sbase which I was given to understand is similar to Azteeg mini, which is based on Smoothie board.

#define X_SERIAL_TX_PIN  P1_22   // J8-2
#define X_SERIAL_RX_PIN  P2_12   // J8-4 Interrupt Capable
#define Y_SERIAL_TX_PIN  P1_23   // J8-3
#define Y_SERIAL_RX_PIN  P2_11   // J8-5 Interrupt Capable
#define Z_SERIAL_TX_PIN  P2_12   // J8-4
#define Z_SERIAL_RX_PIN  P0_25   // TH3
#define E0_SERIAL_TX_PIN P4_28   // J8-6
#define E0_SERIAL_RX_PIN P0_26   // TH4

Problem is, I can't seem to confirm if these pins are the right ones to use.  How does one go about tracing a pin if we don't get have the source info?

ModMike, 2018-08-09:
So after looking at the schematics for V3, it looks like the only controllable thing is VREF.  I will just go stand alone for now.

closed 2018-08-09 by ModMike

kazolar, 2019-08-14:
I found this while investigating a possibility of running uart drivers on x5 mini via marlin. It looks fairly straight forward. You need to make Y cables with 1Kh resistor going to the tx pin and straight to rx pin from uart/pdn pin on the trinamic. Then pick one of the several available pins on the EXP2 header -- best is pair 1.31+3.26  2.11+3.25 1.23+0.17, and 0.16+0.18 -- they can be your software rx/tx pins for the 4 drivers. I am thinking of trying this out -- no reason it shouldn't work.