MarlinFirmware/Marlin #11495: Unable to Flash Update

oralallen82, 2018-08-09:
//file Configuration.h Release:heads/1.1.x
//section info
#define STRING_CONFIG_H_AUTHOR "(Oral Allen, Anet config)"
//section machine
#define BAUDRATE 115200
#define MOTHERBOARD BOARD_ANET_10
#define CUSTOM_MACHINE_NAME "Oral's 3D Printer"
#define POWER_SUPPLY 1
#define INVERT_Y_DIR false
#define INVERT_Z_DIR true
#define X_BED_SIZE 220
#define Y_BED_SIZE 220
#define X_MIN_POS -33
#define Y_MIN_POS -10
#define Z_MAX_POS 240
//section extruder
#define DEFAULT_NOMINAL_FILAMENT_DIA 1.75
#define EXTRUDE_MINTEMP 160
//section temperature
#define TEMP_SENSOR_0 5
#define TEMP_SENSOR_BED 5
#define TEMP_RESIDENCY_TIME 6
#define TEMP_BED_RESIDENCY_TIME 6
#define BED_MAXTEMP 130
#define PID_FUNCTIONAL_RANGE 15
//#define DEFAULT_Kp.0 21.0
//#define DEFAULT_Ki.0 1.25
//#define DEFAULT_Kd.0 86.0
#define BED_LIMIT_SWITCHING 
#define PREHEAT_1_TEMP_HOTEND 190
#define PREHEAT_1_TEMP_BED 60
#define PREHEAT_2_TEMP_BED 90
//section homing
#define X_MIN_ENDSTOP_INVERTING true
#define Y_MIN_ENDSTOP_INVERTING true
#define Z_MIN_ENDSTOP_INVERTING true
#define Z_MIN_PROBE_ENDSTOP_INVERTING true
#define ENDSTOP_INTERRUPTS_FEATURE 
#define HOMING_FEEDRATE_XY (100*60)
//section motion
#define DEFAULT_AXIS_STEPS_PER_UNIT { 100, 100, 400, 100 }
#define DEFAULT_MAX_FEEDRATE { 400, 400, 8, 50 }
#define DEFAULT_MAX_ACCELERATION { 2000, 2000, 100, 10000 }
#define DEFAULT_ACCELERATION 400
#define DEFAULT_RETRACT_ACCELERATION 1000
#define DEFAULT_TRAVEL_ACCELERATION 1000
//section probes
#define PROBE_MANUALLY 
#define X_PROBE_OFFSET_FROM_EXTRUDER 0
#define Y_PROBE_OFFSET_FROM_EXTRUDER 0
#define XY_PROBE_SPEED 6000
//section calibrate
#define AUTO_BED_LEVELING_BILINEAR 
//#define G26_MESH_VALIDATION 
#define RIGHT_PROBE_BED_POSITION 190
#define FRONT_PROBE_BED_POSITION 15
#define ABL_PROBE_PT_1_X 20
#define ABL_PROBE_PT_1_Y 160
#define ABL_PROBE_PT_2_X 20
#define ABL_PROBE_PT_2_Y 10
#define ABL_PROBE_PT_3_X 180
#define ABL_PROBE_PT_3_Y 10
#define LCD_BED_LEVELING 
//section extras
#define EEPROM_SETTINGS 
//#define HOST_KEEPALIVE_FEATURE 
//#define DEFAULT_KEEPALIVE_INTERVAL 2
//section lcd
#define SDSUPPORT 
//#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 2
//#define LCD_FEEDBACK_FREQUENCY_HZ 5000
#define ZONESTAR_LCD 

//file Configuration_adv.h Release:heads/1.1.x
//section temperature
#define THERMAL_PROTECTION_PERIOD 60
#define THERMAL_PROTECTION_HYSTERESIS 10
#define THERMAL_PROTECTION_BED_PERIOD 60
#define THERMAL_PROTECTION_BED_HYSTERESIS 5
#define WATCH_BED_TEMP_PERIOD 180
//#define AUTOTEMP 
//section homing
#define ENDSTOPS_ALWAYS_ON_DEFAULT 
//section extras
//#define ARC_SUPPORT 
//section lcd
//#define SD_DETECT_INVERTED

timeshell, 2018-08-09:
Well, this is so very vague.

oralallen82, 2018-08-09:
Apologies total newbie. Please close this.

closed 2018-09-12 by thinkyhead