MarlinFirmware/Marlin #11496: [1.1.9] Reboot at manual (prepare move axis E) extruding

Tofoillote, 2018-08-09:
### Description

While attempting to move filament via "prepare" "move axis" "Extruder" 10 or 1 or 0.1 mm first step is shown on screen then screen go blue and printer reboot.

Printing Gcode works fine.

### Steps to Reproduce

Load some filament.
Go in prepare "preheat PLA end"
When temp is ok go in "prepare" "move axis" "Extruder" chose a value.
Turn the encoder one direction
value freeze.
wait for reboot.

This configuration is for Creality Ender-3 based on one finded in present Github.

[Marlin_1.1.9_CR_ENDER-3.zip](https://github.com/MarlinFirmware/Marlin/files/2273210/Marlin_1.1.9_CR_ENDER-3.zip)

thinkyhead, 2018-09-12:
Try lowering the `MANUAL_FEEDRATE` for the E axis and see if it helps.

boelle, 2019-02-20:
@Tofoillote did you try what thinky suggested?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

Tofoillote, 2019-03-10:
Yep you can close it. I did assume haven't find the origin, but following flashing did works.

boelle, 2019-03-10:
only you can close it with the close button below

Tofoillote, 2019-03-10:
Thanks, I'm a bit newbie.

closed 2019-03-10 by Tofoillote