MarlinFirmware/Marlin #11497: #define NOZZLE_PARK_FEATURE  WHY IT'S GOING ALL WAY UP ?

bogaskilop, 2018-08-09:
### Description

I just  configured  1.1.9  and set like in 1.1.6 or 1.1.8  part position to 👍 
#define NOZZLE_PARK_POINT { (X_MIN_POS + 314), (Y_MIN_POS ) }

and I wanted to start a print and just after  G28 and G29 it' moving all the way up on Z axis  to XY par_point?

in 1.1.6 or 1.1.8  after G27 it goes on same Z height  to xy from nozzle_park_point and really prefer that 
than moving  all the way up on Z axis before print or over filament change.

bogaskilop, 2018-09-03:
today i got  some time and i downloaded  rc bugfix again  and  still same situation how can i make it to work like in 1.1.6  marlin so  nozzle park position will be  xy whichi set in configuration.h  and Z   on same level or  +5 on Z from current Z position ?

MoellerDi, 2018-09-03:
did you try:
#define NOZZLE_PARK_POINT { (X_MIN_POS + 314), (Y_MIN_POS )**, 5** }

According to the config file a park position as { X, Y, Z }, not sure but it seems you missed Z

bogaskilop, 2018-09-04:
MoellerDi 

yes I wasnt using Z in 1.1.6   and in default 1.1.6  Z stays on same level   when pause was pressed and G27 executed.  becouse when you let say set up my x y park position and add Z 5 mm   and you already with biger model on your build plate  let say you presed pause or want to change filament  and pritn is already  higher  than 5 mm on Z   let say it is alredy on 100  it will crash into model moving to Z 5 :(

thats why I wan to know how to set Z position to stay in same level when pause was executed.

thinkyhead, 2018-09-04:
> it will crash into model moving to Z 5

The Z element is a raise, not a position.

bogaskilop, 2018-09-04:
Ok Thinkyhead i will try this  but please explain me  why it goes all the way up to Z_max if Z is not specified in  park position ?

thinkyhead, 2018-09-05:
With 1.1.9 `NOZZLE_PARK_POINT` requires X, Y, and Z. If no Z is specified, the Z value is random.