MarlinFirmware/Marlin #11498: Delta printer inconsistent homing in marlin bugfix 2.0

luigipotato, 2018-08-09:
### Description
upon homing the sliders sometimes stop halfway and start moving towards the endstops in A, B, C order.
This causes binding because the sliders are still to far away from the endstops.

### Steps to Reproduce

- move nozzle to the bed and do G28
- Do a G33
- Do set delta height from menu
- do G28 several times in home position

**Expected behavior:** ABC move up all together until the endstops

**Actual behavior:** sliders stop half way and starts homing in A then B then C order
[Configuration files.zip](https://github.com/MarlinFirmware/Marlin/files/2274245/Configuration.files.zip)

#### Additional Information

I am coming from firmware Marlin 1.1.0-RC8 without any problems.
this problem also occurs in the current release of marlin.
Movie of the problem: https://1drv.ms/v/s!AjLozuegLiFb5zyf_pseFSNrJexZ

AnHardt, 2018-08-09:
Care about propper noise free endstops.
Look after the new endstop options.

luigipotato, 2018-08-09:
I already soldered a 100nf ceramic cap on the endstops like stated in the configuration file. But it didnt help. Everything worked fine with the alser version of the firmware.

thinkyhead, 2018-11-05:
```cpp
#define ENDSTOP_NOISE_THRESHOLD 2
```
(or higher, as needed)

closed 2018-11-05 by thinkyhead