MarlinFirmware/Marlin #11501: Linear Advance Test Pattern Lines All Look The Same

Splic3dBread, 2018-08-09:
**Expected behavior:** Test pattern lines show variation of thickness between slow and high speed with a certain range displaying an even line.

**Actual behavior:** All test pattern lines across a wide range of K values look identical with barely extruded high speed portion of line.

#### Additional Information

I'm using Linear Advance 1.5 with Marlin 1.1.9. I have experimented with K values ranging from 0 to 12 with 0.1 value stepping and different orientations of the test pattern. The speed settings are left as default and my retraction speed is 55mm/s at 4mm which is what I normally use with great results. The high speed part of all the lines are extremely thin and look like they're barely extruded. The bed is level; extruder and flow rate have been calibrated, however I bumped the flow rate back to 1 from the calibrated value of 0.88 to see what would happen and there was no change. The hotend and extruder are the stock CR-10's with Capricorn tubing.

Here is the gcode from my latest test.
[kfactor.zip](https://github.com/MarlinFirmware/Marlin/files/2275383/kfactor.zip)

cfelicio, 2018-08-12:
can you also post your configuration and configuration_adv files?

Splic3dBread, 2018-08-12:
I'm away from home right now but I can tell you the configuration_adv is stock. I couldn't find a definitive answer to this anywhere online but I gathered that linear advance should be disabled in Marlin for the testing.

ManuelMcLure, 2018-08-12:
> I gathered that linear advance should be disabled in Marlin for the testing.

If you mean disabled by manually setting the K factor to 0 (`M900 K0`) that's fine. But if you mean disabled as in `LIN_ADVANCE` being commented out in `Configuration_adv.h` then no. You must have `LIN_ADVANCE` uncommented in `Configuration_adv.h` for the test to work.

Splic3dBread, 2018-08-13:
Thank you, will try.

closed 2018-08-13 by Splic3dBread