MarlinFirmware/Marlin #11505: Compiling error

makskr, 2018-08-09:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

forkoz, 2018-08-10:
compiling is broken because of platform IO: https://github.com/MarlinFirmware/Marlin/pull/11506

makskr, 2018-08-10:
Thank you forkoz!
I can compile it now.
I have replaced "platform= nxplpc" to "platform = nxplpc@<3.4.0" in platformio.ini

closed 2018-08-11 by makskr