MarlinFirmware/Marlin #11509: [1.1.9] RRD Full Graphics Smat Display shifted

epoxi, 2018-08-10:
This is related to issue #5703 described by @lukeskymuh.
On 1.1.9 the display shows a shifted and/or random pixels. Even using the new LIGHTWEIGHT_UI feature.

On 1.1.8 the LCD works perfect. 

This was tested on 2 different LCD modules from different manufacturers (some mirno PCB changes between them). Both of them works perfectly on 1.1.8 and present the same issue on 1.1.9.

I have no luck playing with the delays :(

Controller: RUMBA
LCD: RRD Full Graphics Smat Display

lokus77, 2018-08-21:
I had same issue with marlin 1.1.9 , and I fixed it! 
My display is #define **REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER**

Same problem: 
  Compile&upload 1.1.8 TFT OK
  Compile&upload 1.1.9 TFT bad.
I tried several versions of U8glib, and clear all arduino cache libraries folder several times. i tried several arduino IDE versions.

At the end.. 
i edited my **ultralcd_st7920_u8glib_rrd.h** and edit delay times for lcd.

my current file has  (lines 43 to 75)

// If you want you can define your own set of delays in Configuration.h
**#define ST7920_DELAY_1 DELAY_NS(50)
#define ST7920_DELAY_2 DELAY_NS(188)
#define ST7920_DELAY_3 DELAY_NS(50)**

**// comented all IF**
//#if F_CPU >= 20000000
//  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
//  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
//  #define CPU_ST7920_DELAY_3 DELAY_NS(50)
//#elif MB(3DRAG) || MB(K8200) || MB(K8400) || MB(SILVER_GATE)
//  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
//  #define CPU_ST7920_DELAY_2 DELAY_NS(188)
//  #define CPU_ST7920_DELAY_3 DELAY_NS(0)
//#elif MB(MINIRAMBO)
//  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
//  #define CPU_ST7920_DELAY_2 DELAY_NS(250)
//  #define CPU_ST7920_DELAY_3 DELAY_NS(0)
//#elif MB(RAMBO)
//  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
//  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
//  #define CPU_ST7920_DELAY_3 DELAY_NS(0)
//#elif MB(BQ_ZUM_MEGA_3D)
//  #define CPU_ST7920_DELAY_1 DELAY_NS(0)
//  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
//  #define CPU_ST7920_DELAY_3 DELAY_NS(189)
//#elif F_CPU == 16000000
//  #define CPU_ST7920_DELAY_1 DELAY_NS(0)

//  #define CPU_ST7920_DELAY_2 DELAY_NS(0)
//  #define CPU_ST7920_DELAY_3 DELAY_NS(63)
//#else
//  #error "No valid condition for delays in 'ultralcd_st7920_u8glib_rrd.h'"
//#endif

Maybe your LCD need another delays. 
( I saw the solution in #4994   )
[image]

thinkyhead, 2018-08-21:
Once you find some delay times that work, please close this issue.

closed 2018-09-09 by thinkyhead