MarlinFirmware/Marlin #11512: [2.0.x] Losing steps when using Firmware Retraction

tcm0116, 2018-08-10:
### Description

Using firmware retraction appears to cause missing steps. I made several attempts at a print that consisted of multiple cubes while using firmware retraction, but each attempt resulted in the cubes becoming skewed towards the origin of the bed on my printer. I finally got the print to succeed by having the slicer perform the retraction instead of using G10/G11.

### Steps to Reproduce

1. Create a GCode file that has multiple cubes and uses G10/G11 commands for retraction.
2. Print the GCode file on a cartesian machine that has Firmware Retraction enabled.
3. The cubes will be skewed towards the origin of the bed.

**Expected behavior:** The cubes are not skewed,

**Actual behavior:** The cubes are skewed.

#### Additional Information

I'm fairly sure this is due to the call to `SYNC_PLAN_POSITION_KINEMATIC()` in `FWRetract::retract()`. My thinking  is that the re-computation of `Planner::position` in `Planner::_set_position_mm()` is causing `current_position` and `Planner::position` to become out of sync, possibly due to rounding errors. I haven't looked into this hypothesis in detail, but it seems like a likely place for steps to get dropped.

thinkyhead, 2018-08-21:
I'm checking over the PR and patching a compile issue:
```
In file included from Marlin/src/module/planner.h:41:0,
from Marlin/src/Marlin.cpp:35:
Marlin/src/module/delta.h: In function 'void inverse_kinematics(const float*)':
Marlin/src/module/delta.h:83:61: error: cannot convert '<brace-enclosed initializer list>' to 'const float*' for argument '1' to 'void inverse_kinematics(const float*)'
inverse_kinematics({raw[X_AXIS], raw[Y_AXIS], raw[Z_AXIS]});
^
In file included from Marlin/src/feature/bedlevel/../../module/planner.h:41:0,
from Marlin/src/feature/bedlevel/bedlevel.cpp:34:
Marlin/src/feature/bedlevel/../../module/delta.h: In function 'void inverse_kinematics(const float*)':
Marlin/src/feature/bedlevel/../../module/delta.h:83:61: error: cannot convert '<brace-enclosed initializer list>' to 'const float*' for argument '1' to 'void inverse_kinematics(const float*)'
inverse_kinematics({raw[X_AXIS], raw[Y_AXIS], raw[Z_AXIS]});
^
```
This approach seems fine, though I wonder how compatible it (and fw-retract in-general) is with the recover from power-loss feature. It currently doesn't save the retract flag, so if power goes out in-between a retract and a recover, the recover won't happen and the Z position might be off by the z-hop amount when the print resumes. Anyway, that feature needs other work too, but it's something to think about going forward.

brucehvn, 2018-09-09:
This may be related to a problem I've been seeing.  I'm seeing my z-axis lowering on every retract/unretract.  It's happening both with firmware retraction and manual retraction.  I've swapped out every piece of hardware involved, ramps board, mega board, stepper driver, LCD display.  I even removed the extruder cable and then the E stepper driver, but can still recreate the problem.  The Z axis motors move for the exact amount of time the extruder motor is moving.  Even if I slow the retract speed down to 4 mm/s2 which is consistent with normal extrusion, it still turns the z motors so it's definitely related to retract/unretract.  The Z motors always turn the same direction regardless of extruder direction.  This is using bugfix-2.0.x from around Aug 26 and I downloaded the latest on Sep 7.

This is a prusa I3 variant running a Mega 2560 R3 and RAMPS 1.4.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/2364098/configs.zip)

thinkyhead, 2018-09-09:
@brucehvn — Try #11578 and see if it solves the issue for you. The branch to download is at https://github.com/tcm0116/Marlin/tree/2.0.x_fwretract

tcm0116, 2018-09-09:
@brucehvn you probably need to explicitly checkout the "2.0.x_fwretract" branch after you clone the repository as that's not the default branch.

brucehvn, 2018-09-10:
@tcm0116 printing with your code I don't see any problem with the z motors moving on retract/unretract.  I'm able to print again with UBL. Yay!

Any idea on when this will be merged into the main bugfix-2.0.x branch?

thinkyhead, 2018-09-10:
@brucehvn — We're just waiting for confirmation that delta motion is still working properly with that PR, and then it will be merged.

studiodyne, 2018-09-13:
Same issue for dual color printing that use a purge bucket for each tool change , that go outside of the bed and back to the object .
Really happy !!!! Good Job !!!!

brucehvn, 2018-10-20:
@tcm0116 @thinkyhead, just checking back to see if this ever got merged.

tcm0116, 2018-10-20:
@brucehvn - yes, it did! I'll go ahead and close this issue.

closed 2018-10-20 by tcm0116