MarlinFirmware/Marlin #11513: 1.1.x Marlin doesnt compile after enabling delta_auto_calibration

jaanshgo, 2018-08-10:
I am trying to enable delta autocalibration but it gives me an error in marlin_main.cpp around line 5669. This is the error:

`sketch/Marlin_main.cpp:5669: undefined reference to `lcd_probe_pt(float const&, float const&)'`

This is the marlin_main.cpp line that causes the error:

`float z_shift = lcd_probe_pt(0, 0) - center;`

The error says `lcd_probe_pt` is undefined but it is defined a little before, here `float lcd_probe_pt(const float &rx, const float &ry);` 
that variable is also called a few lines before the error and it seems to be ok.

Any insight on the error will be appreciated.

jaanshgo, 2018-08-10:
I just tried a fresh version of marlin 1.1.9 as mine is modified quite a bit and I get the same error. This printer has no LCD btw.

jaanshgo, 2018-08-10:
I made a small c++ program and wrote just that variable and I get the same error. It's probably a typo from the developers but I'm not sure what they meant to write so I don't know how to fix it.

jaanshgo, 2018-08-11:
I had to enable something in config.h file. I wish there was a comment mentioning that. I had to dig though a lot of files to find that out.

closed 2018-08-11 by jaanshgo

thinkyhead, 2018-08-12:
@LVD-AC — Due to this issue `G33` currently requires an LCD controller. Can we simply alter this block to remove that dependency, or could there be other LCD dependencies?
```diff
- #if HAS_BED_PROBE
+ #if HAS_BED_PROBE && ENABLED(ULTIPANEL)
    static float probe_z_shift(const float center) {
      STOW_PROBE();
      endstops.enable_z_probe(false);
      float z_shift = lcd_probe_pt(0, 0) - center;
      endstops.enable_z_probe(true);
      return z_shift;
    }
  #endif
```

jaanshgo, 2018-08-12:
lcd_probe_pt gets declared in another file. It gets declared twice for some reason but in that other file it has all the code it is supposed to run. so you would have to modify that other file. I don't remember which one it was but you can find it easily with ctrl f and put lcd_probe_pt

LVD-AC, 2018-08-13:
@thinkyhead You also need to disable the -1 case:

```
          case -1:
          - #if HAS_BED_PROBE
          + #if HAS_BED_PROBE && ENABLED(ULTIPANEL)

              zprobe_zoffset += probe_z_shift(z_at_pt[CEN]);
            #endif

```

thinkyhead, 2018-08-13:
Thanks! That was also included in the patch.

LVD-AC, 2018-08-13:
@thinkyhead see updated comment

thinkyhead, 2018-08-13:
How come that requires an LCD controller?

LVD-AC, 2018-08-13:
not critical but useless without offset calibration

thinkyhead, 2018-08-13:
Ah, I see!