MarlinFirmware/Marlin #11520: [1.1.9] tmc2130: axis not moving when stepper current is set to 1000

jb2304, 2018-08-11:
### Description

My axes stop moving when i set the current to 1000 (standard is 800). With 1.1.8 it worked flawless. I use an MKS_GEN_L Board with TMC2130 drivers on all axes.
With the current set to 800 alls axes home and move as expected. But every axis i set to the current of 1000 does not move at all. 
I flashed 1.1.8 again to make sure its not a hardware Problem. All axis work flawless regardless of the current i set the steppers to.

### Steps to Reproduce

1. Edit configuration_adv.h OR use M906 command to set the stepper current to 1000.
2. Flash firmware OR save setting with M500.
3. Home/move axis with new current.

**Expected behavior:** 
Axis should move and home like usual regardless of the set current.

**Actual behavior:** 
Axis does not move at all and Printer halts when i try to home the axis.

#### Additional Information

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2280276/configuration.zip)

marcio-ao, 2018-08-14:
I suggest enabling TMC_DEBUG and M122. Check to see what the "Run current" is set to. It should be a fraction, like 28/31. Now, compare the values when the current is 800 vs when it is 1000. Make sure that the fraction is larger on the second case.

The reason I suggest this is that in one occasion I had to troubleshoot a similar problem, and it turned out that the "Run current" value was overflowing, causing it to wrap around to a lower value.

thinkyhead, 2018-09-06:
@jb2304 — See if the latest bugfix code is working better for you.

boelle, 2019-02-20:
@jb2304 tried latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-07-11 by shitcreek