MarlinFirmware/Marlin #11521: Question/Discussion LPC17xx SPI, SD, USB future direction

gloomyandy, 2018-08-11:
HI,
I have an MKS SBase setup and have been working away getting 2.0.x working on it and adding a few features that seem of use (eeprom in flash). The current SD card, SPI, USB setup is rather complex and has a few problems and I thought it would be useful to perhaps discus possible future directions. I know that @p3p has been working on updating some things in this area and I'd like to help out if I can. I thought I'd start with describing my understanding of the current setup and potential issues...

1. Built in SD card
This is currently only used to hold system files and can be accessed either by Marlin (to store/load eeprom images) or via USB by a host PC. It is also used to install new firmware. The current implementation makes use of the ChanFS package to provide both low level sector access (for use by the USB MSC interface and as a FAT filesystem (for use by Marlin to store the eeprom image). This package in turn makes use of a hardware SPI interface provided by the CMSIS SPI driver.

2. External SD card (attached to the LCD)
This is normally used to hold G-Code for printing etc. It is accessed by the FAT file system implementation used by the rest of Marlin (located in the src/sd dir). This in turn uses SPI access provided via the HAL layer, both hardware and software implementations are provided.

3. Graphics LCD
This is attached to the board via the EXP1 and EXP2 headers and is driven via a version of the u8g library and this provides an SPI interface to the hardware (both hardware and software interfaces are provided, but I think that it usually uses a software implementation).

On the SBase (and other Smoothieware based boards), the two SD cards share an SPI bus. Currently this does not work (due to several issues including the different SPI implementations being used). So as a work around the SPI connection to the external SD card is routed to a different set of headers by a custom cable.

I hope the above is a reasonably correct version of how things work at the moment, please correct me if I've got things wrong. 

I'm not sure if any/all of these issues are currently being addressed or have been discussed in the past, if so please let me know what I can do to help. Either way I'd like to learn about any previous discussion and contribute if I can. Some of these issues are obviously far reaching and may require some big changes and so may not be worth attempting, I'm sure others can make those calls better than me as I'm not that familiar with the code base. 

Personally I'd love to get rid of the custom cable and free up the pins that are currently used by it (though others may have other priorities as other boards do not have an issue with this). It seems to me that to do this would require at least:
1. Change the current USBMSC code so that the SPI bus is not accessed at interrupt time. My understanding is that some other HALs already do this (like the DUE?).
2. Provide a hardware SPI interface that can address multiple devices.
3. Modify the current HAL SPI interface used by the Marlin SD card code to use this hardware SPI.
Can anyone see other things that are needed? Does anyone see any fundamental issues with doing this?

Finally if this is not the right way to raise a discussion like this then please feel free to close it and point me to a better place!

p3p, 2018-08-11:
As I am moving the framework implementation out of Marlin, most of this will need to be fixed in the framework rather than Marlin, decoupling and moving the framework to a repository was my first step in that direction, once I have that decoupled and merged into (out of) Marlin then I (We) can start working on the Arduino framework and internal libraries, (such as SPI) and making them compatible. 

A lot of the usb and sd card code was only designed to work on the Re-ARM which doesn't share the SPI bus, at the time I only had that board to work with so it does all need reworked.

The SPI and USB are my top priorities on getting sorted out once the framework is split out to https://github.com/p3p/pio-framework-arduino-lpc176x, hopefully will be getting a PR for that in soon, it all works, it's just a big change and don't wan too break too much.

The split is supposed to be the first phase, after that I wan't to move the framework to CMSIS5 so we are not working with ancient code (although I doubt nxp are updating the LPC176x version at all) and use a complete usb library (although lpc176x libusbd isn't really up to date either) rather than the current cobbled together solution.

> Finally if this is not the right way to raise a discussion like this then please feel free to close it and point me to a better place!

Some of the active devs hang out at https://marlinfirmware.slack.com/ but you will need invited by email address.

forkoz, 2018-08-12:
So the NXP updates in platform-io aren't touching the CMSIS code? I see some of it was written in '10 and '11, yikes.

It would be cool if one could use the internal SD for gcode and then fire it off with the USB connection. Drag & Drop from windows/linux and no risk of disconnection/slowdowns plus no getting up. Kind of the best of both worlds.

I know tho that this stuff is probably a long ways away.

tcm0116, 2018-08-19:
>It would be cool if one could use the internal SD for gcode and then fire it off with the USB connection.

There was some discussion about this at one point. @p3p can explain further, but I believe there's an issue where Windows doesn't correctly lock and unlock the filesystem when accessing is via USB. As such, it's not really possible to share the internal SD card with a Windows host and allow Marlin to read gcode files from it.

gloomyandy, 2018-08-19:
Looking at the DUE HAL it looks like that code is setup to share the SD card with a USB host system. It looks like when the printer has a file open (or is printing), then access from the host system is disabled. I suspect that this will work most of the time with no issues, as in typical use the user will copy a file over then start printing it. So no access at the same time. But I can imagine situations which could leave the file system in a corrupt state (in the same way that removing a card mid update might). Is anyone using the DUE in this way, are there problems in reality?

I've been looking at some traces and it seems that Windows will lock the filesystem when making updates to some elements, but it does not keep the system locked while writing the data part of a file. Perhaps this makes sense as it only locks things when making critical changes?

p3p, 2018-08-19:
@tcm0116 is remembering correctly from my trials and tribulations of getting it working "well enough" initially, I had a reasonably nice system that worked for linux but had to rip it out (and try weird workarounds) because Windows didn't seem to follow the standard (there was a lot of staring at USB - SCSI logs and cursing involved). There should be no access to the block device without it locked, read or write, every access is critical.

Sorry I'm taking so long to get the initial work on the framework overhaul completed and a PR submitted, time seems to fly by atm, The plan is to have to mount/unmount between devices.

gloomyandy, 2018-08-19:
Given that I've spent several days doing battle a shared SPI bus combined with Windows USB access (so SPI operations delayed from the USB interrupt to allow shared access to the SPI bus), I can believe that you had a lot of problems! I'm intrigued as to how well the DUE setup is working as it looks like that shares access between USB and the printer.

closed 2019-02-19 by thinkyhead