MarlinFirmware/Marlin #11524: AVR Boot load problems

Roxy-3D, 2018-08-12:
I think I lost the boot loader on my Rumba board.    I used a USBasp and Arduino to put the boot loader back on the board.     I am now once again able to upload code to the board using the normal USB cable and AVR ISP.    

But...  Sadly, the LCD Panel just keeps screeching the buzzer and the uploaded code doesn't seem to do anything.    Are there specific boot loaders I should try?   I'm open to any recommendations that anybody has!!!      Can I use the USBasp to upload the Marlin firmware?

ManuelMcLure, 2018-08-12:
You might want to look at http://forum.arduino.cc/index.php?topic=356420.0 post #5 - there appears to be some issue with the Mega2560 fuses that needs to be resolved before uploading a new bootloader will work.

Roxy-3D, 2018-08-12:
Thank You!     I did verify the Marlin firmware is running.   I can do a M114 and a M119 and get expected numbers.   But NOTHING else works!!!!!    And the stupid LCD Panel just keeps screeching at me.

ManuelMcLure, 2018-08-12:
That behavior seems very odd. Once the boot loader has started Marlin up its job should be done. Unless the boot loader is setting up something wrong on the chip before passing control on to Marlin, which expects a specific state...

Roxy-3D, 2018-08-12:
I only skimmed the links contained in the first link you posted.   I'll dig in deeper tomorrow.   But there is some bug in the Mega-2560 with regard to fuses and being able to see the entire memory space.

If only half of Marlin is there...  That would explain a lot.       (I suspect this theory is way too simple and not correct.   But something like this would explain what is going on.)

igorepi, 2018-08-13:
USBASP with latest official or stock chinese firmware don't work with Atmega chips with more than 128Kb flash, i.e. Atmega2560. In order to program/verify all the Atmega2560 flash you'll need reflash USBASP with modified FW, https://github.com/PeterVH/usbasp/blob/master/bin/firmware/usbasp-v1.5.hex for example. Or you can edit official firmware yourself as [in this commit](https://github.com/PeterVH/usbasp/commit/14af55e03c5e4f5238d8eec6f83f8360405a7193): this single edit will be sufficient.

The problem is explained [here](https://petervanhoyweghen.wordpress.com/2015/12/02/the-usbasp-and-atmega2560-mystery/).

boelle, 2019-03-13:
@Roxy-3D should this one still be open?

closed 2019-05-16 by thinkyhead