MarlinFirmware/Marlin #11528: char signedness issue in ultralcd_impl_HD44780.h

nikwrt, 2018-08-12:
### Description

When using a 16x2 LCD the _draw_heater_status(const int8_t heater, const char prefix, const bool blink) function in ultralcd_impl_HD44780.h is called with prefix=-1. Unfortunately this does not work because the Makefile sets -funsigned-char, meaning char is by default unsigned. As a result, prefix is an unsigned char and -1 is interpreted as 255 by the function, which makes it prints extra unwanted characters.   

### Steps to Reproduce

1. Compile release 1.1.9 after configuring for a 16x2 LCD such as MAKEBOARD_MINI_2_LINE_DISPLAY_1602

**Expected behavior:** [What you expect to happen]
Short top line with extruder temperature/bed temperature and no symbols

**Actual behavior:** [What actually happens]
Unwanted extra characters before each temperature

### Proposed fix
Change declaration to _draw_heater_status(const int8_t heater, const **signed** char prefix, const bool blink)

thinkyhead, 2018-08-15:
Thanks for the report! I'm not sure why we have the `-funsigned-char` flag set. I've always assumed we could trust `char` to be signed. And there's no code in Marlin that operates on the assumption that `char` is always `unsigned`. So, we might just drop that flag from the `Makefile`.

The `Makefile` is only used for building from the command-line… so is that how you're doing the build? Or do you also see this when building in Arduino / PlatformIO?

closed 2018-08-15 by thinkyhead

thinkyhead, 2018-08-15:
Marlin expects `signed` historically, so we should add `-fsigned-char` to ensure this.

p3p, 2018-08-15:
That is why I did add it, but the ARM ABI is unsigned for any precompiled system libraries and linking to them with `-fsigned-char` causes issues (which is why I mentioned it)

thinkyhead, 2018-08-21:
I see! As I'm trying to catch up on ~300 issues in the queue, if you or someone else wants to produce a more thorough solution it would be great to have a PR for it. I simply don't have time to scrub the entire codebase to touch every `char` reference right now.