MarlinFirmware/Marlin #11529: STM32F407VET6 and STM32F103RCT6 it reboot when going home

semos2k, 2018-08-12:
### Description

I am installing marlin 2.0.x on the boards: STM32F407VET6 and STM32F103RCT6 at the time of homing, the board reboot.

### Steps to Reproduce

1. tested NO and NC
#define X_MIN_ENDSTOP_INVERTING false // set to true to invert the logic of the endstop.
#define Y_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MIN_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define X_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Y_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define Z_MAX_ENDSTOP_INVERTING true // set to true to invert the logic of the endstop.
#define ENDSTOP_INTERRUPTS_FEATURE (active and inactive)

**Expected behavior:** Homing

**Actual behavior:** reboot board, does not work on any of the axes.

AnHardt, 2018-08-12:
Are you sure your endstops are not shorting 5V to ground?

semos2k, 2018-08-12:
yes, they work properly with an arduino board.

thinkyhead, 2018-11-05:
Working with the latest `bugfix-2.0.x`?

AnHardt, 2018-11-05:
Rebooting while moving or at the moment an endstop is hit?
Reboot when endstop is manually triggered - without movement?

boelle, 2019-02-20:
@semos2k tried latest bugfix 2.0 ?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead