MarlinFirmware/Marlin #11532: While G28: Z moves up and up with every X move

ikarisan, 2018-08-12:
I just flashed the most recent 2.0.x-bugfix version ( aa9d329560eb9c8de2dd68bbf0797c200e540b33 ) and I have a homing issue. On every X move the Z axis moves up. After a second G28 I ended up near below Z max. :( 

See this video: https://youtu.be/pz5_J92eIMY 

With an older version (around 07. August) everything worked fine.
I am using TMC2208 for X and Y on a MKS Gen 1.4 board.

ikarisan, 2018-08-14:
After doing some further investigations I can clearly say, that 0af22549a6780d3486da895be7a920bd6c02927f is the last working version. Beginning from c93538e9f3d409c9b96921d0d9ed4681fa0226b6 (UBL/planner patch tweaks) the homing is broken.

YMita2017, 2018-08-16:
Same situation but bugfix-1.1.x.
From 1.1.7 to 1.1.9, same.
Use same setting from 1.17 to 1.1.9.
At autohoming, or G28, always Z moves up and up.
At 1.1.7, Z move up before X homing, but Z not move untill Y homing finish.
I'm using Precision Piezo.

Graham-3D, 2018-08-16:
I am having the same issue.  I installed [bugfix-2.0.x] from a day or so back (not sure of the version) and my printer is behaving the same as in the video. I reverted back to 1.1.9 and everything works correctly again. I also have a BLTouch (Well 3D Touch). Identical settings copied over as far as I can tell.

smoki3, 2018-08-16:
Same here @thinkyhead

Graham-3D, 2018-08-20:
My issue here has been fixed by [2.0.x] Fix UBL/Homing #11566

thinkyhead, 2018-09-09:
Assuming everything is configured correctly, BLTouch should work fine with 1.1.9 and 2.0.x.

boelle, 2019-02-20:
@ikarisan tried latest bugfix 2.0 ?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead