MarlinFirmware/Marlin #11534: MKS_MINI_12864 and MLS_GEN_L and Marlin bugfix 2

RudyBenner, 2018-08-13:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

RudyBenner, 2018-08-13:
Cannot compile with MKS Mini 12864. Will compile with the 'standard' Reprap discount full graphic smart controller, however, when flashed, display is corrupt.
Have also tried with and without sdsupport.
[MKS mini 12864.zip](https://github.com/MarlinFirmware/Marlin/files/2281665/MKS.mini.12864.zip)

thinkyhead, 2018-08-14:
It compiles fine now using your configs with the latest `bugfix-2.0.x` code. Whatever was causing the issue seems to have been patched.

thinkyhead, 2018-08-14:
I'm able to compile with your configs (`MKS_MINI_12864`) using the latest `bugfix-2.0.x` code. So whatever was causing the compile error seems to have been patched.

RudyBenner, 2018-08-14:
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from sketch\src\inc/MarlinConfig.h:37:0,

                 from sketch\src\Marlin.h:25,

                 from sketch\src\Marlin.cpp:31:

sketch\src\inc/SanityCheck.h:1474:4: error: #error "Please select no more than one LCD controller option."

   #error "Please select no more than one LCD controller option."

    ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Atatoth, 2018-08-14:
You set more than one LCD in config

RudyBenner, 2018-08-14:
I see that now, my bad. Here is the error message with Arduino beta.

Arduino: 1.9.0-beta (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

In file included from c:\windows\temp\arduino_build_989527\sketch\src\hal\hal_avr\hal.h:38:0,

                 from c:\windows\temp\arduino_build_989527\sketch\src\inc\marlinconfig.h:28,

                 from sketch\src\lcd\ultralcd.h:26,

                 from sketch\src\lcd\ultralcd.cpp:29:

c:\windows\temp\arduino_build_989527\sketch\src\pins\pins_RAMPS.h:513:38: error: pasting "/* backlight LED on A11/D65*/" and "_DDR" does not give a valid preprocessing token

       #define LCD_BACKLIGHT_PIN 65   // backlight LED on A11/D65

                                      ^

c:\windows\temp\arduino_build_989527\sketch\src\hal\hal_avr\fastio_AVR.h:84:42: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)

                                          ^

c:\windows\temp\arduino_build_989527\sketch\src\hal\hal_avr\fastio_AVR.h:102:35: note: in expansion of macro 'SET_OUTPUT'

 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)

                                   ^

sketch\src\lcd\ultralcd_impl_DOGM.h:299:5: note: in expansion of macro 'OUT_WRITE'

     OUT_WRITE(LCD_BACKLIGHT_PIN, HIGH);

     ^

sketch\src\lcd\ultralcd_impl_DOGM.h:299:15: note: in expansion of macro 'LCD_BACKLIGHT_PIN'

     OUT_WRITE(LCD_BACKLIGHT_PIN, HIGH);

               ^

c:\windows\temp\arduino_build_989527\sketch\src\pins\pins_RAMPS.h:513:38: error: pasting "/* backlight LED on A11/D65*/" and "_PIN" does not give a valid preprocessing token

       #define LCD_BACKLIGHT_PIN 65   // backlight LED on A11/D65

                                      ^

c:\windows\temp\arduino_build_989527\sketch\src\hal\hal_avr\fastio_AVR.h:84:61: note: in definition of macro '_SET_OUTPUT'

 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)

                                                             ^

c:\windows\temp\arduino_build_989527\sketch\src\hal\hal_avr\fastio_AVR.h:102:35: note: in expansion of macro 'SET_OUTPUT'

 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)

                                   ^

sketch\src\lcd\ultralcd_impl_DOGM.h:299:5: note: in expansion of macro 'OUT_WRITE'

     OUT_WRITE(LCD_BACKLIGHT_PIN, HIGH);

     ^

sketch\src\lcd\ultralcd_impl_DOGM.h:299:15: note: in expansion of macro 'LCD_BACKLIGHT_PIN'

     OUT_WRITE(LCD_BACKLIGHT_PIN, HIGH);

[... 273 lines cut ...]

c:\windows\temp\arduino_build_989527\sketch\src\hal\hal_avr\fastio_AVR.h:102:51: note: in expansion of macro 'WRITE'

 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)

                                                   ^

sketch\src\lcd\ultralcd_impl_DOGM.h:299:5: note: in expansion of macro 'OUT_WRITE'

     OUT_WRITE(LCD_BACKLIGHT_PIN, HIGH);

     ^

sketch\src\lcd\ultralcd_impl_DOGM.h:299:15: note: in expansion of macro 'LCD_BACKLIGHT_PIN'

     OUT_WRITE(LCD_BACKLIGHT_PIN, HIGH);

               ^

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

RudyBenner, 2018-08-14:
I will download a new distribution file and compile.

RudyBenner, 2018-08-14:
leaving off most of it....

"C:\Users\RudolfBenner\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-gcc-ar" rcs  "C:\WINDOWS\TEMP\arduino_build_14174\core\core.a" "C:\WINDOWS\TEMP\arduino_build_14174\core\new.cpp.o"
Archiving built core (caching) in: C:\WINDOWS\TEMP\arduino_cache_412230\core\core_arduino_avr_mega_cpu_atmega2560_32485795a840bdcd62af6a6560d52731.a
Linking everything together...
"C:\Users\RudolfBenner\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -w -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections,--relax -mmcu=atmega2560  -o "C:\WINDOWS\TEMP\arduino_build_14174/Marlin.ino.elf" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\Marlin.ino.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\usb\ctrl_access.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\usb\sysclk.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\usb\udc.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\usb\udi_cdc.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\usb\udi_cdc_desc.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\usb\udi_composite_desc.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\usb\udi_msc.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\usb\uotghs_device_due.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\usb\usb_task.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\include\digipot_mcp4451_I2C_routines.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\u8g\HAL_LCD_I2C_routines.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\u8g\HAL_LCD_pin_routines.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\Marlin.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\persistent_store_api.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_AVR\HAL.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_AVR\HAL_spi_AVR.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_AVR\MarlinSerial.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_AVR\persistent_store_eeprom.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_AVR\persistent_store_impl.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_AVR\servo_AVR.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_AVR\watchdog_AVR.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\DebugMonitor_Due.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\EepromEmulation_Due.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\HAL.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\HAL_spi_Due.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\HAL_timers_Due.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\InterruptVectors_Due.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\MarlinSerialUSB_Due.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\MarlinSerial_Due.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\Servo_Due.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\Tone.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\persistent_store_eeprom.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\persistent_store_impl.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_shared_hw_spi.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\u8g_com_HAL_DUE_st7920_sw_spi.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\watchdog_Due.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_DUE\usb\sd_mmc_spi_mem.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_ESP32\HAL.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_ESP32\HAL_spi_ESP32.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_ESP32\HAL_timers_ESP32.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_ESP32\ota.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_ESP32\watchdog_ESP32.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\HAL.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\HAL_spi.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\HAL_timers.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\LPC1768_PWM.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\LPC1768_Servo.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\SoftwareSPI.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\WInterrupts.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\arduino.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\main.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\persistent_store_flash.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\persistent_store_sdcard.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\u8g_com_HAL_LPC1768_hw_spi.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\u8g_com_HAL_LPC1768_ssd_hw_i2c.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LPC1768\u8g_com_HAL_LPC1768_st7920_hw_spi.cpp.o" "C:\WINDOWS\TEMP\arduino_build_14174\sketch\src\HAL\HAL_LP [... cut ...] toa.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\libraries\U8glib\clib\u8g_u8toa.c.o" "C:\WINDOWS\TEMP\arduino_build_14174\libraries\U8glib\clib\u8g_virtual_screen.c.o" "C:\WINDOWS\TEMP\arduino_build_14174/core\core.a" "-LC:\WINDOWS\TEMP\arduino_build_14174" -lm
avr-gcc: error: CreateProcess: No such file or directory

Using library U8glib at version 1.19.1 in folder: C:\Users\RudolfBenner\Documents\Arduino\libraries\U8glib 
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

RudyBenner, 2018-08-14:
Arduino: 1.8.5 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

Archiving built core (caching) in: C:\WINDOWS\TEMP\arduino_cache_520730\core\core_arduino_avr_mega_cpu_atmega2560_32485795a840bdcd62af6a6560d52731.a
fork/exec C:\Users\RudolfBenner\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-gcc.exe: The filename or extension is too long.
Error compiling for board Arduino/Genuino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

RudyBenner, 2018-08-14:
No go, also tried the Arduino 9 beta. No traction. 1.1.9 compiles. Also now cannot compile WITHOUT the MKS_MINI_12864. No LCDs selected.

RudyBenner, 2018-08-14:
Fresh start. New folder, freshly unzipped Marlinbugfix2.0. The default config compiles, as would be expected. 
That would be the RAMPS motherboard. 
Make one (1) change, from this //#define MKS_MINI_12864 to #define MKS_MINI_12864

Does not compile.
I need a large refreshment.

AnHardt, 2018-08-14:
Two well known problems.
Marlin bugfix-2.0.x can't be compiled with Arduino IDE below beta because of too long paths during compilation - will not find files.
And bugfix-2.0.x will currently not compile with Arduino beta because of a bug where inline comments // in #defines are not well recognised. Solution is, either to remove the comments where you get errors or to use PlatformIO for the compilation. (We strongly believe in a fix by Arduino before beta will be released - for that we do not remove/replace the comments for now. )

RudyBenner, 2018-08-14:
Interesting. I get same results with Arduino IDE beta. I will put aside my current working config file and edit out the comments as you have suggested.

RudyBenner, 2018-08-14:
No resolution.

thinkyhead, 2018-08-14:
> Make one (1) change, from this `//#define MKS_MINI_12864` to `#define MKS_MINI_12864`

I just tested this (compiling with PlatformIO for mega2560) and got no errors.

RudyBenner, 2018-08-15:
I installed Marlin-Conf again, this time I get success. Some months ago, it led me to grief. Hard to argue with results. Thanks for your help and patience.

RudyBenner, 2018-08-15:
Unfortunately, I am only a legend in my mind. When I was younger, I could make stuff work that others abandoned. Now I am 70 and I am not sure I should be allowed to work on a toaster. I still try. Often I get lucky, just often enough to maintain the dream.
Rudy (ex electrician)

closed 2018-08-15 by RudyBenner