MarlinFirmware/Marlin #11536: Issues with  Marlin 1.1.8 firmware.

nhatsunflower, 2018-08-13:
### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

nhatsunflower, 2018-08-13:
Hi everyone !
I'm building a delta pritnter with Marlin 1.1.8 firmware. 
I have met some issues and I lost more than 3 weeks to fix it but it still didn't work>
When using the auto calibration on the printer. All steps in delta auto calibaration works perfectly then I store setting and I see everything about delta settings on LCD is changed. But when I try to print or move X,Y,Z axis. the center is always perfect. then it acts like a rainbow from side to side of x and y. Autotune fails. Manual tune fails. Never saves the settings! I always hit store and they show but it's like it ignores them. 
Please help me !

thinkyhead, 2018-08-13:
Do you definitely have `EEPROM_SETTINGS` enabled?

boelle, 2019-02-20:
@nhatsunflower tried what thinky suggested? using bugfix 2.0?

closed 2019-02-20 by thinkyhead