MarlinFirmware/Marlin #11541: Marlin 1.1.9 not reporting E0 TMC2130 driver

daverdfw, 2018-08-14:
### Description

With Marlin 1.1.9 and TMC2130 installed for X,Y,Z,E0 using the SW SPI I can get status for all doing an M122 except for E0
### Steps to Reproduce

1. Run M122

**Expected behavior:** [What you expect to happen]
Should see list of all drivers including E0
**Actual behavior:** [What actually happens]
Not seeing E0 driver listed in M122 output

Recv: Enabled		true	true	true
Recv: Set current	750	800	750
Recv: RMS current	734	795	734
Recv: MAX current	1035	1121	1035
Recv: Run current	23/31	25/31	23/31
Recv: Hold current	11/31	12/31	11/31
Recv: CS actual		23/31	25/31	23/31
Recv: PWM scale		168	130	177
Recv: vsense		1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true
Recv: msteps		16	16	16
Recv: tstep		675	272	1048575
Recv: pwm
Recv: threshold		0	0	0
Recv: [mm/s]		-	-	-
Recv: OT prewarn	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false
Recv: off time		5	5	5
Recv: blank time	24	24	24
Recv: hysteresis
Recv: -end		2	2	2
Recv: -start		3	3	3
Recv: Stallguard thrs	16	16	0
Recv: DRVSTATUS	X	Y	Z
Recv: stallguard
Recv: sg_result		0	0	0
Recv: fsactive
Recv: stst
Recv: olb			X	X
Recv: ola			X	X
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x80:17:00:00
Recv: 	Y = 0x40:19:00:00
Recv: 	Z = 0x60:17:00:00

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2285141/configuration.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#how-can-i-contribute) for additional guidelines.

teemuatlut, 2018-08-14:
[TMC troubleshooting section on Marlin documentation site](http://marlinfw.org/docs/hardware/tmc_drivers.html#troubleshooting)
(For you: update to the bugfix branch)

daverdfw, 2018-08-14:
Ok I will try the bugfix branch. Any idea if in 1.1.9 there was an issue with the hardware SPI ? I could never get it to work until I went with the SW SPI. Thanks !

teemuatlut, 2018-08-14:
I don't think so. The very core communication for the TMCs hasn't changed much or at all for quite a while and I haven't seen any reports that would indicate fundamental issues like that. Of course we regularly have users reporting communication issues but those are very hard to diagnose remotely and can be due to a multitude of hardware problems.
SW SPI option was never meant to replace the HW implementation, but rather to make it easier to use the TMC drivers with an LCD.

daverdfw, 2018-08-14:
Ok thanks for confirming. I followed the guide put out by a user here on github. And clearly the wiring works because it worked with the SW. kinda weird it never would register when using the HW pins.

daverdfw, 2018-08-15:
Ok, updated to latest 1.1.x-bugfix and now I see E0! I had to do a M502 to wipe out the old power settings that M122 would report. Thanks!

closed 2018-08-15 by daverdfw