MarlinFirmware/Marlin #11543: [EEPROM functions unusable for RAMPS-FD v2] problem of persistent_store_eeprom.cpp in 2.0.x Aug 14 build

craftoscar, 2018-08-14:
### Description

EEPROM functions unusable for RAMPS-FD v2 on 2.0.x-Bugfix Aug 14 build 

I'm migrating my older July 7th 2.0.x-bugfix to newer 2.0.x-Bugfix Aug 14 build

And find out there is some problem in  `persistent_store_eeprom.cpp` DUE HAL

error code form Arduino IDE and platform IO are simmilar
`sketch\src\HAL\HAL_DUE\persistent_store_eeprom.cpp: In static member function 'static bool PersistentStore::write_data(int&, const uint8_t*, size_t, uint16_t*)':

sketch\src\HAL\HAL_DUE\persistent_store_eeprom.cpp:43:14: error: decrement of read-only parameter 'size'

   while (size--) {

              ^

sketch\src\HAL\HAL_DUE\persistent_store_eeprom.cpp: In static member function 'static bool PersistentStore::read_data(int&, uint8_t*, size_t, uint16_t*, bool)':

sketch\src\HAL\HAL_DUE\persistent_store_eeprom.cpp:70:14: error: decrement of read-only parameter 'size'

   } while (--size);

              ^

sketch\src\HAL\HAL_DUE\persistent_store_eeprom.cpp: In static member function 'static const size_t PersistentStore::capacity()':

sketch\src\HAL\HAL_DUE\persistent_store_eeprom.cpp:87:10: error: 'E2END' was not declared in this scope

   return E2END + 1;

          ^

exit status 1
Error compiling for board Arduino Due (Programming Port).
`

I fund out there is some change from `persistent_store_impl.cpp `(July 7th) to `persistent_store_eeprom.cpp`(Aug 14th)

I already isolate other errors by simply, chang   `#define MOTHERBOARD BOARD_RAMPS_FD_V2` 

and enabling EEPROM wright functions by uncommon `#define EEPROM_SETTINGS` in `configuration.h`

Is there are some modification of `persistent_store_eeprom.cpp` in progress or am I missed out some library?

>  persistent_store_impl.cpp(July 7th):

`/**
 * Marlin 3D Printer Firmware
 *
 * Copyright (C) 2016 MarlinFirmware [https://github.com/MarlinFirmware/Marlin]
 * Copyright (c) 2016 Bob Cousins bobcousins42@googlemail.com
 * Copyright (c) 2015-2016 Nico Tonnhofer wurstnase.reprap@gmail.com
 * Copyright (c) 2016 Victor Perez victor_pv@hotmail.com
 *
 * This program is free software: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License as published by
 * the Free Software Foundation, either version 3 of the License, or
 * (at your option) any later version.
 *
 * This program is distributed in the hope that it will be useful,
 * but WITHOUT ANY WARRANTY; without even the implied warranty of
 * MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 * GNU General Public License for more details.
 *
 * You should have received a copy of the GNU General Public License
 * along with this program.  If not, see <http://www.gnu.org/licenses/>.
 *
 */
#ifdef ARDUINO_ARCH_SAM

#include "../persistent_store_api.h"

#include "../../inc/MarlinConfig.h"

#if ENABLED(EEPROM_SETTINGS)

extern void eeprom_flush(void);

bool PersistentStore::access_start() { return true; }

bool PersistentStore::access_finish() {
  #if DISABLED(I2C_EEPROM) && DISABLED(SPI_EEPROM)
    eeprom_flush();
  #endif
  return true;
}

bool PersistentStore::write_data(int &pos, const uint8_t *value, const size_t size, uint16_t *crc) {
  while (size--) {
    uint8_t * const p = (uint8_t * const)pos;
    uint8_t v = *value;
    // EEPROM has only ~100,000 write cycles,
    // so only write bytes that have changed!
    if (v != eeprom_read_byte(p)) {
      eeprom_write_byte(p, v);
      if (eeprom_read_byte(p) != v) {
        SERIAL_ECHO_START();
        SERIAL_ECHOLNPGM(MSG_ERR_EEPROM_WRITE);
        return true;
      }
    }
    crc16(crc, &v, 1);
    pos++;
    value++;
  };
  return false;
}

bool PersistentStore::read_data(int &pos, uint8_t* value, const size_t size, uint16_t *crc, const bool writing/*=true*/) {
  do {
    uint8_t c = eeprom_read_byte((unsigned char*)pos);
    if (writing) *value = c;
    crc16(crc, &c, 1);
    pos++;
    value++;
  } while (--size);
  return false;
}

bool PersistentStore::write_data(const int pos, uint8_t* value, const size_t size) {
  int data_pos = pos;
  uint16_t crc = 0;
  return write_data(data_pos, value, size, &crc);
}

bool PersistentStore::read_data(const int pos, uint8_t* value, const size_t size) {
  int data_pos = pos;
  uint16_t crc = 0;
  return read_data(data_pos, value, size, &crc);
}

const size_t PersistentStore::capacity() {
  return E2END + 1;
}

#endif // EEPROM_SETTINGS
#endif // ARDUINO_ARCH_SAM
`

> `persistent_store_eeprom.cpp(Aug 14th)`

`#ifdef ARDUINO_ARCH_SAM

#include "../persistent_store_api.h"

#include "../../inc/MarlinConfig.h"

#if ENABLED(EEPROM_SETTINGS)

extern void eeprom_flush(void);

namespace HAL {
namespace PersistentStore {

bool access_start() {
  return true;
}

bool access_finish(){
#if DISABLED(I2C_EEPROM) && DISABLED(SPI_EEPROM)
  eeprom_flush();
#endif
  return true;
}

bool write_data(int &pos, const uint8_t *value, uint16_t size, uint16_t *crc) {
  while (size--) {
    uint8_t * const p = (uint8_t * const)pos;
    uint8_t v = *value;
    // EEPROM has only ~100,000 write cycles,
    // so only write bytes that have changed!
    if (v != eeprom_read_byte(p)) {
      eeprom_write_byte(p, v);
      if (eeprom_read_byte(p) != v) {
        SERIAL_ECHO_START();
        SERIAL_ECHOLNPGM(MSG_ERR_EEPROM_WRITE);
        return true;
      }
    }
    crc16(crc, &v, 1);
    pos++;
    value++;
  };
  return false;
}

bool read_data(int &pos, uint8_t* value, uint16_t size, uint16_t *crc, const bool writing/*=true*/) {
  do {
    uint8_t c = eeprom_read_byte((unsigned char*)pos);
    if (writing) *value = c;
    crc16(crc, &c, 1);
    pos++;
    value++;
  } while (--size);
  return false;
}

}
}

#endif // EEPROM_SETTINGS
#endif // __AVR__
`

salfter, 2018-08-14:
This appears to be a regression introduced in commit 66d2b48b5995a4225029f7fed84a2dfe0ee9300a.  In my case, Marlin won't even build:

```
Marlin/src/HAL/HAL_DUE/persistent_store_eeprom.cpp: In static member function 'static const size_t PersistentStore::capacity()':
Marlin/src/HAL/HAL_DUE/persistent_store_eeprom.cpp:65:10: error: 'E2END' was not declared in this scope
return E2END + 1;
^
Marlin/src/HAL/HAL_DUE/persistent_store_eeprom.cpp:66:1: warning: control reaches end of non-void function [-Wreturn-type]
}
^
*** [.pioenvs/DUE/src/src/HAL/HAL_DUE/persistent_store_eeprom.cpp.o] Error 1
```

60f13767985da920cf77197a6441f6aaad57a2a8 is, AFAICT, the last commit that builds for RAMPS-FD.

craftoscar, 2018-08-14:
@salfter 
thanks for your replay 
I'm not quite a coder, so can't really understand it well

> Marlin won't even build
the same situation here

marcio-ao, 2018-08-14:
I am getting the same situation compiling for the Archim 2.0 board.

marcio-ao, 2018-08-14:
Upon further investigation, the commit which broke the build for me was 66d2b48b5995a4225029f7fed84a2dfe0ee9300a

p3p, 2018-08-14:
Sorry about this, appears we need an extra test or 2 for the DUE platform, it's getting hard to update HAL apis without knowing (and having test hardware for) all platforms.

There are a few issues, a later merge appears to have duplicated some renamed files, the include in persistent_store_eeprom.cpp changed from `#include "../../inc/MarlinConfig.h`". But one thing I don't know is where E2END actually gets defined in the DUE platform ...

p3p, 2018-08-14:
Indeed that's how it currently is, any board with an onboard spi/i2c EEPROM needs to define E2END, the problem was (along with a missing header) that the default Flash emulated EEPROM code doesn't seem to define it. How the spi/i2c overrides work needs to be cleaned up to use the PersistentStore class rather than just redefining the avr arduino functions.

marcio-ao, 2018-08-14:
Are you guys saying that on an Archim 2.0 (which is a 32-bit board like the Due), Marlin is going to need to access the SPI FLASH chip for configuration? In the prototype I am developing I decided to store a startup animation on the SPI FLASH because I did not think Marlin was using it and I was free to use that storage for whatever. This discussion concerns me a bit...

marcio-ao, 2018-08-14:
Could someone explain what types of memory are available on an ARDUINO_ARCH_SAM? Does it have EEPROM? I enabled EEPROM_SETTINGS and it seems to work alright (i.e. M500 and M502 seem to work). The earlier comments that the Due does not have EEPROM confuses me, as then I don't really understand where this data is being stored.

The Archim 2.0 board does have an SPI FLASH chip, but I assumed that this storage was independent and unused by Marlin. I have been using it for my own purposes. So far I have seen no conflicts between that and the Marlin M500 and M502.

p3p, 2018-08-14:
@marcio-ao Marlin can access SPI EEPROMs for configuration but doesn't have to, the Due platform also seems to supports storing the configuration in the Flash (EepromEmulation_Due.cpp), this is used by default when SPI_EEPROM and I2C_EEPROM are disabled.

marcio-ao, 2018-08-14:
@p3p: Ah, I see. So as long as keep SPI_EEPROM and I2C_EEPROM disabled I can continue to use the SPI FLASH chip on the Archim for my own uses, is that so? I assume that this means the M500/M502 will be stored on on-board flash of the SAM chip itself and that this is the purpose of EepromEmulation_Due?

If that is the case, I suspect that is what I want.  One thing I don't quite understand is how come my build fails looking for E2END? It sounds like maybe that code isn't being wrapped with the proper #ifdefs? Should that code only be active if "SPI_EEPROM" or "I2C_EEPROM" are enabled?

p3p, 2018-08-14:
> So as long as keep SPI_EEPROM and I2C_EEPROM disabled I can continue to use the SPI FLASH chip on the Archim for my own uses, is that so?

Indeed shouldn't be a problem

> One thing I don't quite understand is how come my build fails looking for E2END? 

I'm fixing that now along with some other issues that were not picked up in the tests (or happened in the merge), E2END is a constant usually defined by the AVR headers when a chip has EEPROM, in the case of the DUE it needs to be defined by whatever is supplying the EEPROM but the EEPROM Flash emulation doesn't seem to be doing so.

marcio-ao, 2018-08-14:
> E2END is a constant usually defined by the AVR headers when a chip has EEPROM, in the case of the DUE it needs to be defined by whatever is supplying the EEPROM but the EEPROM Flash emulation doesn't seem to be doing so.

@p2p: Ah. Gotcha.

BTW, I was just looking through "EepromEmulation_Due" ... whoever put that together *really* went through great lengths to do wear leveling. I had assumed Flash chips did their own wear leveling internally (or is that just SD cards and/or USB thumbdrives?)

p3p, 2018-08-14:
Wear levelling is handled by the an mcu in those cases, Flash chips are just dumb blocks of memory. The program memory in mcus is also not designed for as much continuous rewrite as a memory stick would be.

salfter, 2018-08-16:
I'm not so sure everything's resolved yet.  I added this to pins_RAMPS_FD_V2.h:

```
#define E2END 0xFFFF
```

Marlin builds to completion once again, but actual EEPROM access appears to be broken.  It won't read the settings that had previously been stored there, and attempts to write to it result in "Error writing to EEPROM!" messages.  Would it be better to reopen this issue or create a new one?

p3p, 2018-08-16:
Can you try `#define E2END 0xFFF` Marlin doesn't usually handle such large EEPROMs, although that shouldn't stop it reading it, we haven't changed anything in the i2c eeprom code and that doesn't use it (only UBL does), I assume it functions if you comment out the `I2C_EEPROM` and use the flash eeprom version of the code?.

salfter, 2018-08-16:
I tried that, and 0x1FFF...made no difference either way.  It does work if I disable I2C_EEPROM, so it would appear that gremlins are still lurking in that part of the code.

I just rolled back to what I thought was the last version that worked (last upstream commit in my branch is 60f13767985da920cf77197a6441f6aaad57a2a8), but it's behaving the same way if I2C_EEPROM is enabled.  When I first got this RAMPS-FD up and running, f891416e701ccd9e6153c20d55ac9407a0fcf017 was current...maybe I need to roll it back to that and see  what happens.

salfter, 2018-08-16:
...and it looks like the EEPROM wasn't working even with https://github.com/MarlinFirmware/Marlin/commit/f891416e701ccd9e6153c20d55ac9407a0fcf017.  Looks like a new issue is probably in order.

salfter, 2018-08-16:
...and I just realized something: I've been trying to test EEPROM functionality while my RAMPS-FD was unplugged from my Arduino.  The EEPROM was disconnected!

#include <picard_facepalm.h>

salfter, 2018-08-16:
...and as soon as I plugged in the RAMPS-FD and loaded up the most recent commit with I2C_EEPROM enabled and E2END set to 0xFFFF, the last-saved settings were pulled in.  I issued M502 followed by M500; they worked as expected.

I'll go sit in the corner now, wearing the dunce cap.

p3p, 2018-08-16:
Glad you got it sorted, we all make silly mistakes now and then ^^