MarlinFirmware/Marlin #11547: Bugfix 2.0 - Servo pin problem

J-charles-C, 2018-08-14:


J-charles-C, 2018-08-14:
I,

If i switch ServoPin (E0_AUTO_FAN_PIN P1_1**8** & FAN_PIN P1_1**9**) 
E0_AUTO_FAN_PIN works fine and turn on when hotend is hot but Layer fan don't turn on.

I can't use two ServoPin at the same time.

kAdonis, 2018-08-14:
Your Problem could be the IRF 520, this Mosfet is not made for 3,3V devices like the Re-Arm
Pins 2_04, 2_05, 2_07 (D8, D9 and D10 on RAMPS)  and P1_18 (Servo) are level shifted to 5V, which is enough for an IRF 520 (VGS(th)=2-4V). 
But the other Pins on the servo Header deliver only 3,3V which **might be** not enough for your IRF 520 to switch.
You could try a logic-level-Mosfet for example an IRLZ34N, I found this by googling, I am no expert in electronics, there will be better choices for sure.
The SevenSwitch from your #11545 should work also (uses MOSFET IRLB 8743)
Or your issue is caused by something completely different

J-charles-C, 2018-08-14:
I,

I have used this bridge for 5V to servo :
[image]

kAdonis, 2018-08-14:
That doesn't matter. I was talking about the voltage level on the signal pin itself.
 Pin1_18 5V 
 Pin 1_19 3,3V
 the other Servo pins also 3,3V

J-charles-C, 2018-08-14:
Thanks for your help.

I connected with a Y the 5V from the RESARM to the VCC pin and 5V from the RAMPS.

I just measured, I have 4.6 volts on the input of IFR520.

best.

kAdonis, 2018-08-14:
Sorry, I don't understand what you mean.

kAdonis, 2018-08-14:
[image]
Don't connect 5V to VCC_Pin on RAMPS! You will probably damage your RE-Arm!

kAdonis, 2018-08-14:
I understand you are delivering  5V to the middle pins on the servo connector, this is fine
But this doesn't change the output voltage on the servo-signal pins, you don't need the 5V from the servo header for a Mosfet.
I still don't know what you mean with "Y"

J-charles-C, 2018-08-14:
I still don't know what you mean with "Y"

It's parallel wiring. One point to two point.

best

kAdonis, 2018-08-14:
You really connected 5V to the VCC-Pin on the Picture above?
I just checked the Re-Arm design files, there is no protection diode, this connection directly feeds 5V to the LPC
I can't believe this is working, the MCU must be boiling hot

J-charles-C, 2018-08-15:
yes i do.

It's source of my problem ?

kAdonis, 2018-08-15:
I am suprised that it works at all, but your Re-Arm is probably damaged
Unplug the VCC-Pin!
maybe the electronics inside the chip for pin 1_19 and 1_20 is destroyed, I can't help you with that

The LCD needs 5V, you need a special cable, look here:
http://panucattdevices.freshdesk.com/support/solutions/articles/1000243195-lcd-display-installation

J-charles-C, 2018-08-15:
I have test a switch of the Pin in Ramps.h and with the same effect pin_ramps_re_arm.h

#define SERVO1_PIN          6
#define SERVO2_PIN          4  // Origin 5
#ifndef SERVO3_PIN
  #define SERVO3_PIN        5  // Origin 4

The fan runs connected to the SERVO3_PIN while the addresses are inverted with the SERVO2_PIN.

best

kAdonis, 2018-08-15:
I am not sure , if i understand you
I don't know why you are changing the  pin defines
Pin 1_18 is physically connected to the Servo pin to the right , no matter what you define

[image]

and pins_RAMPS.h is not even used by Marlin when you use a Re-Arm

> Dammaged on Ramps or Re-Arm ?

I think your Re_Arm is damaged

I cannot help you

J-charles-C, 2018-08-15:
Ok, I thought I could redistribute the servo ports.

I modified Ramps.h because it serves as support in the Re-Arm. the file pin_ramps_re_arm.h send back to him.

How to connect the BL-touch to another port that Servo0pin, to release my two servopin that still work?

Thanks for your help.

kAdonis, 2018-08-15:
Is your BLToch connected to Servo1 Pin? What is on Servo0 pin

J-charles-C, 2018-08-15:
Sorry  my bl-touch is on Servo0 pin (D11)

 Servo1 pin &  Servo2 pin are out of service

 Servo3 pin is for my fan layer.

J-charles-C, 2018-08-16:
Hello,

Thanks again for your help.

I found a solution by remapping one of my IRF520 on an enstop.

Like that, I order my two fans and my BL touch.

I'll have to look at this damaged card problem.

cordially

closed 2018-08-16 by J-charles-C

J-charles-C, 2018-08-25:
I, 

i change Ramps and i solve my problem of servo pin out of service.

Many thanks kAdonis
cordially