MarlinFirmware/Marlin #11550: [bugfix-2.0.x] HOMING_Z_WITH_PROBE wrong with BLTouch

berezovskyi-oleksandr, 2018-08-14:
I have problem with Z-homing using BLTouch (actually 3DTouch, but that doesn't matter now) on separate `Z_MIN_PROBE_PIN` pin.

In `Conditionals_LCD.h` there exist next define
```
#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN))
```

I have undefined `Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN` due to separate `Z_MIN_PROBE_PIN` available (board RURAMPS4D_13) and because of that `HOMING_Z_WITH_PROBE` is false. 

That brings us to `Marlin/src/module/motion.cpp:1325` which states:
```
  #if HOMING_Z_WITH_PROBE && ENABLED(BLTOUCH)
    // BLTOUCH needs to be deployed every time
    if (axis == Z_AXIS && set_bltouch_deployed(true)) return;
  #endif
```
And because of that, BLTouch never gets deployed during Z-horming.

I think there should be something like that 
```
#define HOMING_Z_WITH_PROBE (HAS_BED_PROBE && Z_HOME_DIR < 0 && (ENABLED(Z_MIN_PROBE_ENDSTOP) || ENABLED(Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN)))
```
After this change Z-homing started working

thinkyhead, 2018-09-12:
When `Z_MIN_PROBE_ENDSTOP ` is set, Marlin assumes that a regular endstop exists on the Z min plug, and that this endstop is being used for homing. If you want to override this behavior and always use the probe for homing, you can simply set `HOMING_Z_WITH_PROBE` to true in `Conditionals_LCD.h`.

boelle, 2019-02-20:
@berezovskyi-oleksandr is problem still there with latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

berezovskyi-oleksandr, 2019-03-15:
@boelle , sorry for the delay. I've checked yesterday and problem is still there. Latest code still relies on endstop for Z-homing. Fix from my starting message still works pretty well.

StevenP94, 2019-03-15:
@berezovskyi-oleksandr I made the same into my https://github.com/MarlinFirmware/Marlin/issues/13202
ATM I can't issue a G28 without that

boelle, 2019-06-22:
missing label

closed 2019-07-11 by thinkyhead