MarlinFirmware/Marlin #11553: TMC2130 sensorless_homing fails after cancelling print

truglodite, 2018-08-15:
### Description

Bug Report

### Steps to Reproduce

1. Start a print using host (octoprint)
2. While printer is moving, cancel print from host... g-code stops and cancel script follows like this.
3. M84 ;disable steppers
4. M104 S0 & M140 S0 ;disable hotend and hotbed
5. M106 S0 ;disable fan (...this is the end of the cancel script)
6. Then I send G28 to home all

**Expected behavior:**
After G28 is sent, z raises 5mm to clear stuff, x moves home, y moves home, then x and y move to the z_safe_probing spot, z lowers until the inductive probe triggers over a probe target near the left front corner (the target is a piece of aluminum tape on top of the glass bed).

**Actual behavior:**
After G28 is sent, z raises 5mm, but x doesn't home all the way left instead it moves just a hair, then similarly y moves just a hair, the carriage then appears to not move for a z_safe_homing position but instead starts to z lower at wrong spot for probing. The end result is the probe never triggers since it's not over a target, which allows the nozzle to crash into the bed. I have to quickly kill marlin to stop my z motors from grinding away and damaging the nozzle.

#### Additional Information
If I force this failure, a marlin reboot gets everything working again. Sensorless homing and stallguard for x and y are working perfectly for me aside from this issue after cancelling a print. I have homed for maintenance and let the printer sit longer than DEFAULT_STEPPER_DEACTIVE_TIME. After that homing still seems to work as it should. So it doesn't look like the M84 in my cancel script is causing this.

The closed issue #8890 is different in that G28 only worked once for him, and I can do many G28's or G29's in a row without issue... until AFTER I cancel the print. I figured I'd mentioned it since it seems related.

My config is pretty standard; RAMPS_EEB with stealthchop enabled and no modifications to spreadcycle thresholds. I do have some of pins my pins swapped around from defaults due my setup and having a bad Mosfet, but I don't see how this would cause a problem. This problem exists for me regardless which marlin branch I try (both current 1.1.x and 2.0.x bugfix). My attached configs & pins files are for 2.0.x since I'm currently using 2.0.x files I downloaded a couple of days ago.

[configs.zip](https://github.com/MarlinFirmware/Marlin/files/2288968/configs.zip)

Thairay, 2018-08-20:
i got the same issue but I do not use the diag pins of my tmc2130. 
Most of the time the x carrier will crush into the max position if this error occurs.

Thairay, 2018-08-24:
Its like it is saving the coordinates at the ending position of g28 instead of at the endstops position. But only occurs if a print was cancelled.

truglodite, 2018-11-06:
It's been a while, but I haven't experienced anymore occurrences of this issue. I presume a marlin or tmc driver update fixed the problem. So I'm closing this issue.

closed 2018-11-06 by truglodite