MarlinFirmware/Marlin #11561: [bufix-2.0.x latest] DUE+RAMPS-FD V2 DEACTIVATE_SERVOS_AFTER_MOVE cause extruder auto fans not working

scott0122, 2018-08-15:
#11537

I'm using SWITCHING_NOZZLE.
When I enable DEACTIVATE_SERVOS_AFTER_MOVE , extruder auto fans stop working after server.detach() called.
But when I disable DEACTIVATE_SERVOS_AFTER_MOVE , extruder auto fans work fine.

So I enable DEACTIVATE_SERVOS_AFTER_MOVE and modify src\HAL\servo.cpp. Extruder auto fans works.
Is extruder auto fans share the same time with servo?

```
void Servo::detach() {
  servo_info[this->servoIndex].Pin.isActive = false;
  /*
  timer16_Sequence_t timer = SERVO_INDEX_TO_TIMER(servoIndex);
  if (!isTimerActive(timer)) finISR(timer);
  */
```
}

thinkyhead, 2018-11-05:
Solved?

scott0122, 2018-11-09:
@thinkyhead 

No, still not working.

boelle, 2019-02-20:
@scott0122 tried latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-04-17 by scott0122