MarlinFirmware/Marlin #11562: [bufix-2.0.x latest] UBL keep probing fail after 6th or 8th and above probe point...

Ccleorina87, 2018-08-16:
I trying the [bufix-2.0.x latest] and have problem UBL (G29 P1) with some probing point fail. i revert back to my previous backup which date Aug 13 and it probing perfectly. 

And im also notice when doing G28 the z axis keep going down even though i set to 

#define Z_CLEARANCE_DEPLOY_PROBE        2
#define Z_CLEARANCE_BETWEEN_PROBES   2
#define Z_CLEARANCE_MULTI_PROBE           2 
#define Z_AFTER_PROBING                            0

but after done G28 it did return to 0 point.

smoki3, 2018-08-16:
Looks like the home and leveling code is broken actually. The z axis is always moving and every home and leveling move.

smoki3, 2018-08-16:
https://github.com/MarlinFirmware/Marlin/issues/11532

dmitry-kutergin, 2018-08-16:
The issue is happening on G28 as well and a one move (G0/G1) after it, it happens only if `Z_HOMING_HEIGHT` is not zero.
The symptoms are that Z step are reset to 0 after each axis homing, while maintaining absolute position values. Here is the debug output from `G28 X`, notice steps value for the 3rd move `(0->1000)`:
```
>>> G28
Machine Type: Cartesian
Probe: FIX_MOUNTED_PROBE
Probe Offset X:-34 Y:0 Z:-3.60 (Left-Center & Below Nozzle)
Auto Bed Leveling: UBL (enabled)
Z Fade: 10.00
UBL Adjustment Z 0.000 ( 0.000)
echo:Active Extruder: 0
  current_position=(0.00, 0.00, 0.00) : bracket_probe_move
> endstops.enable(true)
Raise Z (before homing) to 5.00
>>> do_blocking_move_to(0.00, 0.00, 5.00)
  buffer_segment FR:10.00 X:0.00 (0->0) Y:0.00 (0->0) Z:5.00 (0->1000) E:0.00 (0->0)
  buffer_segment FR:200.00 X:0.00 (0->0) Y:0.00 (0->0) Z:5.00 (1000->1000) E:0.00 (0->0)
<<< do_blocking_move_to
>>> homeaxis(X)
Home 1 Fast:
>>> do_homing_move(X, -442.50, [50.00])
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
  buffer_segment FR:50.00 X:-442.50 (0->-35400) Y:0.00 (0->0) Z:5.00 (0->1000) E:0.00 (0->0)
echo:busy: processing
<<< do_homing_move(X)
>>> set_axis_is_at_home(X)
For X axis:
 home_offset = 0.00
 position_shift = 0.00
 soft_endstop_min = 0.00
 soft_endstop_max = 295.00
> home_offset[X] = 0.00
  current_position=(-5.00, 0.00, 5.00) : 
<<< set_axis_is_at_home(X)
  current_position=(-5.00, 0.00, 5.00) : sync_plan_position
  current_position=(-5.00, 0.00, 5.00) : > AFTER set_axis_is_at_home
<<< homeaxis(X)
  current_position=(-5.00, 0.00, 5.00) : sync_plan_position
  current_position=(-5.00, 0.00, 5.00) : bracket_probe_move
echo:Active Extruder: 0
X:-5.00 Y:0.00 Z:5.00 E:0.00 Count X:-400 Y:0 Z:0
<<< G28
```
Here is the 3rd move line when everything is working as it should:
```
>>> do_homing_move(X, -442.50, [50.00])
  current_position=(0.00, 0.00, 5.00) : sync_plan_position
  buffer_segment FR:50.00 X:-442.50 (0->-35400) Y:0.00 (0->0) Z:5.00 (1000->1000) E:0.00 (0->0)
``` 
The root cause is in planner.cpp in a method `void Planner::_set_position_mm(const float &a, const float &b, const float &c, const float &e) `
Line 2549 is missing parenthesis around `#if` clause after `c +`, and due to the operator precedence `position[C_AXIS]` is always `0` steps if UBL is enabled : 
```  
position[C_AXIS] = LROUND(axis_steps_per_mm[C_AXIS] * (c + 
    #if !IS_KINEMATIC && ENABLED(AUTO_BED_LEVELING_UBL)
      leveling_active ? ubl.get_z_correction(a, b) :
    #endif
    0
  ));
``` 
Please commit the following fix, it works fine after that:
```
  position[C_AXIS] = LROUND(axis_steps_per_mm[C_AXIS] * (c + (
    #if !IS_KINEMATIC && ENABLED(AUTO_BED_LEVELING_UBL)
      leveling_active ? ubl.get_z_correction(a, b) :
    #endif
    0)
  ));
```

smoki3, 2018-08-16:
thanks this works great

smoki3, 2018-08-16:
I created two pull requests to fix it.

Ccleorina87, 2018-08-16:
Ok thanks...

Roxy-3D, 2018-08-16:
> I created two pull requests to fix it.

These are merged...   And I'll be bringing up UBL on a new machine over the next few days.   So if there are any problems...   Let me know.   I'll be watching for any issue.

Ccleorina87, 2018-08-16:
Ok Roxy

Ccleorina87, 2018-08-16:
So far i already test this problem and its working perfectly... Thanks guys... i will close this issues

closed 2018-08-16 by Ccleorina87