MarlinFirmware/Marlin #11565: UBL uncommented creates homing bump issue

fr33dimension, 2018-08-16:
I flashed 1.1.9 yesterday and got an unexpected issue. When flashing it with UBL as bedleveling selected (uncommented) homing x and y makes a z move during the bump. So for example, when homing x only, it lifts z 10mm which is good, x travels home hits the stop and now it bumps back and z will move up, both when moving away and back, so a total of 20mm plus the 10mm which are normal. 

It doesn't matter if UBL is active or not, once it's flashed, this behavior starts, if commented out, it does not.
I tried 2.0. as well, same thing, had to go back to 1.1.4

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/2293594/Configuration.zip)

smoki3, 2018-08-16:
i thinks its the same as 
https://github.com/MarlinFirmware/Marlin/issues/11562

smoki3, 2018-08-16:
try this:
https://github.com/MarlinFirmware/Marlin/pull/11567

Roxy-3D, 2018-08-16:
This is merged!