MarlinFirmware/Marlin #11568: Unable to load settings EEPROM Marlin Plugin

RogerHman, 2018-08-16:
Im a noob who has run out of ideas! Any help is appreciated. 

I have upgraded my CR-10 with autobed lvl. I want to change z-offsett via plugin or control box but cant do either.

I have checked the forum and i have all these settings enabeled.

#define EEPROM_SETTINGS // Enable for M500 and M501 commands
#define DISABLE_M503    // Saves ~2700 bytes of PROGMEM. Disable for release!
#define EEPROM_CHITCHAT   // Give feedback on EEPROM commands. Disable to save PROGMEM.

SDSUPPORT is off to save space.

I'm running Marlin 1.1.8. but same problem with 1.1.6

AnHardt, 2018-08-16:
The EEPROM code was changed for 1.1.9 and even more for bugfix-1.1.x .
We will not touch 1.1.6 nor 1.1.8 any more.
Reporting bugs for those old versions doesn't make any sense.
Test bugfix-1.1.x or bugfix-2.0.x.

What do you mean with 'plugin'? 
What do you mean with 'control box'?

What 'forum'? This is a bug tracker. Not a beginners forum.

Please - next time fill the issue form or at least read it.

RogerHman, 2018-08-17:
wrong version

closed 2018-08-17 by RogerHman