MarlinFirmware/Marlin #11569: [FR]Prompt to print with info

sarf2k4, 2018-08-16:
Hi,

I've been meaning to open this idea here and I will try to explain here as much as I can.

**Problem**
1. Print estimates only shows after generating gcode or toolpath in slicer software or upon exporting them. These print time were saved inside the gcode file as well but the print estimate for slic3rpe, were put way at the end of the gcode file. Adding a print estimate onto filename is not what I would say good idea because it will make the filename too long and might make the printer laggy or slower in listing the gcode files. Other printer or firmware, might not have scrolling long gcode filenames.

2. Printing from sdcard, although we might not notice or encounter it, I had issues where I have 3 gcode files; a,b,c files. If i wanted to print file b, either file a or c might get printed instead of b because of a bit of problems with the knobs. The issues here actually prompt the user if the user really want to print the file or not.

3. Types of material used for that particular gcode also saved inside gcode and chances are, it is low to identify them because the filament profile also put at the end of gcode file same as no.1 above. Usually the user would just open the gcode and check the initial print temperature. Same with no.2, filename would becoming too long if its added as well

**Proposal**
Add a new gcode command (probably easier to remember) that accepts strings and the firmware knows that '/n' or '\n' is a newline type or will automatically line break if no '/n' or '\n' specified and too long for a single line.
This new gcode will be generated from slic3r at the very top of the gcode, preferably in "start g-code" in slic3r or any other slicer programs. The slic3r's start g-code variable would then be broaden instead of those allowed variables in the start g-code only.

The printer or the firmware's behavior that I can think of is only simple, There will be a "yes" and "no" selection. If the user picked yes, the printer continues reading the gcode file from the sdcard. If no selected, the printer/firmware immediately closes the gcode file and do nothing. The printer shouldn't cache or buffer any "future" or next line if this specific gcode were to be called. 

This proposed gcode command are as follows:

Example gcode command on one of the generated gcode file
`M909 3dbenchy.gcode\nGeneric PLA\n1H 40M 29S\nHB55' HE210'`

Would translate into the following in the 2004 LCD panel
```
3dbenchy.gcode
Generic PLA
1H 40M 29S
HB55 HE210
```

Then there will be a next page/screen to prompt the user
```
Do you want to proceed with print?
>YES      NO      REVIEW
```

Between the info generated by M909 (for example) and the prompt screen, the firmware/printer will alternate in about 5-10 seconds and user are able to skip the info screen (M909) by pressing the knob to proceed to the prompt screen if it is lesser than the transition delay. 

I hope the above statements are clear.

P.S:
To be honest, a closed source company, Zortrax has this feature which I found that it is helpful to have this extra information.

thinkyhead, 2018-08-21:
It's not very likely we'll add code to Marlin to interpret comments by slicers. But it would be good to give clear feedback on which file is selected for printing.

sarf2k4, 2018-08-22:
what about a special character to indicate it is the end of that particular code's command?

This may introduce buggy feature due to file structure I think

InsanityAutomation, 2019-10-15:
@boelle @shitcreek the confirmation dialog satisfies this as far as id take it.

closed 2019-10-15 by boelle