MarlinFirmware/Marlin #11571: SD Card Not Working.

ingybing, 2018-08-16:
### Description

I've got a 16GB SanDisk Class A1 Micro SD Card in an adapter and a GEEETECH Prusa I3 Pro B printer. The original firmware that came with the printer (v1.0.2) worked just fine with this SD Card, no problems at all. However I upgraded to 1.1.9 and it fails to recognise the card. It sees a card is inserted but the directory listing is blank. I've reformatted with SD Card Formatter, repartitioned to a 1GB partition and reformatted. Intestering once I did that from Windows it lists the files on the card for a second. Then won't let you select any. Instead fills each file line with '/' characters.. then goes back to being blank again with nothing to select.

If i use a old card 4gb class 2 that a bit battered in places (so not sure how much longer it will last hence me wanting to get the sandisk up and running again) it works just fine so. I've tired all the SD Card speed options in the firmware config file to see if it was to do with that but no joy. Something between 1.0.2 and 1.1.9 has broken the support for this SD card type..

### Steps to Reproduce

Pre-Req
1x SanDisk Ultra 16GB Class A1 Micro SD Card
1x Micro SDCard to SD Card Adapter

Steps to re-create
Insert card
Choose Print From SD Card.

**Expected behavior:** 

File should be listed and selectable for printing.

**Actual behavior:** 

Unable to select a file. As in the description either no files are listed or if small partition and windows format the files list briefly but are un selectable. then get replaced with '/' chars on each line, then the files aren't listed any more.

#### Additional Information

[Archive.zip](https://github.com/MarlinFirmware/Marlin/files/2295255/Archive.zip)

boelle, 2019-02-20:
@ingybing still not working if using latest bugfix 2.0?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-17 by p3p