MarlinFirmware/Marlin #11572: [bugfix-2.0.x] UBL/Z-Offset seems to be completely ignored with BLTouch

ikarisan, 2018-08-16:
Hi!

I am using am MKS Gen1.4, BLTouch and the latest marlin bugfix-2.0.x version and it seems tha the UBL and/or homing data is completely ignored. 

I heated up to 75°C bed and 220°C nozzle. After that (waited for 5 minutes) I did a "G29 P1" and stored it to the mesh. After that I tried to print the generated "Linear Advance" pattern but the filament did not stick to the bed. Than I printed the G26 mesh... I tried. But the mesh is printed much to high! I modified Z offset from -1.797 (with this value and G1 Z0 I can hardly put a piece of paper under the nozzle!) to -1.862 but it made things worse. NONE of the printed lines sticked to the bed!

I I home Z with the offset of -1.862, on "G1 Z0" it rams the nozzle into the bed. So the value of -1.797 seems to be the right one.  But not for prints. O.o 

And it seems that the Z offest is implemented the wrong way. The higher I set this value to a negative value (nearer to bed) the worst the print will get! :( 

See  the photos for both Z offsets...

[image]
[image]

ManuelMcLure, 2018-08-16:
What does your mesh look like (output of `G29 T`)? If you disable UBL (`G29 D`) and try to print the `G26` pattern, does it work better?

If you disable UBL, home, and move Z to 0 (`G1 Z0`) and check the distance with a piece of paper, does it still seem like changing the offset from -1.797 to -1.862 makes the nozzle appear further from the bed?

ikarisan, 2018-08-16:
I did two "G29 P1" directly one after the other and these are the (machine usable, G29 T1) results:

23:29 
======
-0.206	-0.169	-0.139	-0.123	-0.034	-0.036	-0.090	-0.071	-0.082	-0.086
-0.210	-0.160	-0.153	-0.129	-0.099	-0.095	-0.097	-0.094	-0.075	-0.081
-0.235	-0.192	-0.172	-0.127	-0.115	-0.100	-0.101	-0.096	-0.070	-0.057
-0.228	-0.212	-0.169	-0.127	-0.106	-0.087	-0.073	-0.065	-0.070	-0.066
-0.231	-0.193	-0.152	-0.123	-0.101	-0.083	-0.090	-0.061	-0.050	-0.029
-0.245	-0.207	-0.158	-0.134	-0.114	-0.090	-0.074	-0.069	-0.056	-0.033
-0.234	-0.205	-0.163	-0.140	-0.110	-0.096	-0.074	-0.061	-0.062	-0.017
-0.226	-0.219	-0.163	-0.139	-0.127	-0.125	-0.088	-0.067	-0.047	-0.010
-0.218	-0.219	-0.163	-0.138	-0.127	-0.125	-0.088	-0.067	-0.032	-0.003
-0.210	-0.219	-0.163	-0.137	-0.127	-0.125	-0.088	-0.067	-0.017	0.004

23:40
======
-0.199	-0.156	-0.125	-0.110	-0.095	-0.078	-0.072	-0.063	-0.072	-0.088
-0.193	-0.182	-0.139	-0.114	-0.103	-0.091	-0.083	-0.079	-0.065	-0.065
-0.220	-0.181	-0.142	-0.120	-0.100	-0.098	-0.095	-0.084	-0.075	-0.065
-0.210	-0.184	-0.153	-0.109	-0.090	-0.071	-0.066	-0.060	-0.066	-0.049
-0.220	-0.175	-0.141	-0.110	-0.084	-0.067	-0.073	-0.052	-0.057	-0.025
-0.222	-0.178	-0.144	-0.116	-0.100	-0.073	-0.062	-0.058	-0.043	-0.022
-0.207	-0.182	-0.153	-0.129	-0.103	-0.091	-0.072	-0.058	-0.041	-0.014
-0.215	-0.190	-0.153	-0.127	-0.114	-0.105	-0.077	-0.067	-0.041	-0.000
-0.215	-0.190	-0.153	-0.125	-0.114	-0.105	-0.077	-0.067	-0.041	0.014
-0.215	-0.190	-0.153	-0.123	-0.114	-0.105	-0.077	-0.067	-0.041	0.028

I will test the "G29 D" in a few minutes...

ManuelMcLure, 2018-08-16:
Hmmm... it looks like the left side of your bed is lower than the right side. Did you manually adjust the bed screws to make all corners even before running the `G29 P1`? If you did, it looks like you might have a twisted X-axis which is making one side of the bed read consistently higher than the other side.

ikarisan, 2018-08-16:
Okay... disabling UBL and printing a G26 results in printing into the air.
Aber disabling UBL, G28 and G1 Z0 there is a large gap between nozzle and bed. :(

I measured the distance (to my frame/Y axis) of the left and the right side of my X axis and it is within the tollerance of the values I am able to measure by hand (+/- 0.2mm). 

Never had these kind of problems a few month ago... :(

ManuelMcLure, 2018-08-16:
Although a tilted X axis (what you measured) would result in one side of the bed appearing higher than the other, what I meant was a twisted X axis - i.e. the printhead rotates around the X axis as it moves from side to side as if it were on a screw (albeit one with an extremely steep pitch). Unless the sensor is at the same Y offset as the nozzle this can cause the `G29 P1` measurements to appear as if the X axis was tilted.

In any case, it's always a good idea to get the bed as close to level as possible using paper at all four corners before trying to use automatic bed leveling. The less the correction needed the better.

ManuelMcLure, 2018-08-16:
Also, do you have commit db306507311862946a966e006a626dfc28f69ad6 in your bugfix-2.0.x?

ikarisan, 2018-08-17:
Hi!

I still don't fully understand what you mean by a twisted axis. :D

Both X axis have been replaced by some high precision ones a few weeks ago. Also all Chinese bearings were replaced by MISUMI bearings. 

I releveled the bed on all four corners and executed "G29 P1". And this is the result (see picture). Will try a G26 later. 

[image]

ikarisan, 2018-08-17:
Okay. Does not make any difference!
I have noticed the following:

Before I print I set my Z offset to -1.6xx. Than I print witg G26 and it fails. If I now call "G28 Z; G1 Z0" the nozzle is far too way from the bed. So I need to set the Z offset to -1.7x. Next try with G26 results in the same desaster. But now, after a "G28 Z" I need to set the offset to -1.8x... respectively I try. But now Marlin is no longer reacting on M851 commands. I have to call G28 to reanimate the printer.

ikarisan, 2018-08-17:
Hmmmm... that sounds logical. But how can the problem be solved? You would need some kind of "x twist" parameter.

ManuelMcLure, 2018-08-17:
What I did was manually level the bed as well as I could using the corner screws, then did a `G29 P0` to zero out the mesh, and just used `G26` and `M421` to manually edit the mesh until `G26` gave a good result. Then I saved the mesh with `G29 S0` and just load it with `G29 L0` before printing. I also printed out [Anet A8/Prusa i3 - Z-Axis Alignment Tool](https://www.thingiverse.com/thing:2465458) which I use any time I might have put strain on the X axis with the motors turned off (like when changing filament) and only use the sensor for homing.

The whole idea of UBL is that you only need to generate the mesh once.

ManuelMcLure, 2018-08-17:
One solution is to find a sensor mount that puts the sensor directly in line with the nozzle - i.e. neither in front or in back of it. Although this doesn't eliminate the error entirely, it reduces it by at least an order of magnitude. A sensor that uses the actual nozzle (a piezo or electrical contact sensor) totally eliminates the error caused by the X axis twist.

Roxy-3D, 2018-08-18:
Yeah...   But because of the Y offset...  the twist gets amplified.   (As your math shows!)

Roxy-3D, 2018-08-19:
@AnHardt  has a good graphic from OpenScad that shows the twist problem.     Maybe he will see this message and post it?

lrpirlet, 2018-08-20:
Have a look at #9529 entry Feb 13th by @AnHardt... Then read the whole topic… A possible solution is given (Y belt tension must be enough but not too much)...

ikarisan, 2018-08-20:
The next thing I will do, just to be sure, is to flash an older bugfix-1.1.x version (around July) and test the UBL. You say that is an hardware issue but as said, the UBL worked with the direct extruder and an old version. 

And why can I set the Z offset at X110/Y110 to a value that a sheet of paper fits only very tight beneath the nozzle and the bed and if I print with this offset, the nozzle seems to print into the air? 

I.e. at 110/110 I measured an Z offset of -1.750mm but even if I set the Z offset to -1.850 for the print the nozzel is far away from the bed.

gloomyandy, 2018-08-20:
Are you sure your homing is producing consistent results? The mesh that you posted earlier is odd in that it seems to contain nearly all negative values, this would imply that you are homing on a spot on the bed that is the high/low point which is pretty unlucky. Where on the bed do you home to? It is probably best to home to the centre of the bed (assuming that is the area you normally use for printing). If you home say six times and after each home perform the paper test do you get the same Z reading each time? If not then something may be wrong with your homing (are you using the same probe for homing and bed levelling?).

ikarisan, 2018-08-20:
I am homing at "bedsize/2" (center of the bed).
Just printed a 40x40x10 cube and it was okay - not good for the first layer, I had to lower Z offset during the print. After that i thought I could print another model. But only the prime line was "okay". The object, printed at the center of the bed, did not stick to it because the gap was to large... again :( 

And yes, I am using a BLTouch Smart for leveling and homing. And the repeatability test gives variations around 0.01mm for the same point. After I reset my printer the offset is okay.. somehow... I had to lower the Z offset again.

ikarisan, 2018-08-24:
I took your clues seriously and modified my X carriage to hold the BLTouch 35mm right of the nozzle (Y=0) instead of 55mm behinde the nozzle. :) 

The UBL mesh results seems to be okay, now. 
The screenshot looks weired, because the corners of the bed must be leveled again. :) 
But in X-direction the deviations are now very small.

ManuelMcLure, 2018-08-24:
I'm glad my comments helped!

boelle, 2019-02-20:
@ikarisan problem still there?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead