MarlinFirmware/Marlin #11573: [bufix-2.0.x latest]  MAX6675  compiling error for DUE

xarwell, 2018-08-16:
When I enable support of MAX6675 i have following compile error.
The same error occurs for MAX31855.
I use board (with TMC2130 SPI) based on arduino due.

      In file included from sketch\src\module\temperature.cpp:37:0:

      sketch\src\module\../libs/private_spi.h: In static member function 'static uint8_t SPIclass<74u, 75u, 76u>::receive()':

      sketch\src\module\../libs/private_spi.h:50:7: error: 'SPDR' was not declared in this scope

       SPDR = 0;

       ^

      In file included from sketch\src\module\thermistor/../../inc/MarlinConfigPre.h:28:0,

                 from sketch\src\module\thermistor/../../inc/MarlinConfig.h:26,

                 from sketch\src\module\thermistor/thermistors.h:26,

                 from sketch\src\module\temperature.h:30,

                 from sketch\src\module\temperature.cpp:27:

      sketch\src\module\../libs/private_spi.h:51:19: error: 'SPSR' was not declared in this scope

       for (;!TEST(SPSR, SPIF););

                   ^

      sketch\src\module\thermistor/../../inc/../core/macros.h:71:23: note: in definition of macro 'TEST'

       #define TEST(n,b) !!((n)&_BV(b))

                       ^

      sketch\src\module\../libs/private_spi.h:51:25: error: 'SPIF' was not declared in this scope

       for (;!TEST(SPSR, SPIF););

                         ^

      sketch\src\module\thermistor/../../inc/../core/macros.h:70:21: note: in definition of macro '_BV'

       #define _BV(n) (1<<(n))

                     ^

      sketch\src\module\../libs/private_spi.h:51:14: note: in expansion of macro 'TEST'

       for (;!TEST(SPSR, SPIF););

              ^

      exit status 1

JackDaFlow, 2018-08-17:
Hey, the "../libs/private_spi.h" isn't working for DUE. I used  #include <SPI.h> instead. But you have to change the code in the temperature.cpp. You can find a working version for the MAX31865 in my rep.

tig33r, 2018-08-18:
Correct me if I am wrong but DUE does not have properly working soft spi implementation? Right now situation is clear, sd card works with hardware spi and nothing more can work with hardware spi. TMC drivers are working with soft spi implemented in @teemuatlut library.

JackDaFlow, 2018-08-18:
Hey,

I started with Soft SPI  for my TMC's and LCD on my DUE. Of course the LCD and TMCs  share the same SW MOSI MISO SCK. Now I'm switched to HW-SPI. Had no issue with HW and SW SPI. I never tested the MAX31865 for SW-SPI. 
Anyway, I newer ran SW and HW SPI in parallel. Actually I tried, but It never worked properly (on DUE and AVR). So as long you stick to one SPI type, you should be ok. just be sure that they are running with the same settings. (I updated my settings, now they all running on the same SPI_SPEED).

tig33r, 2018-08-18:
Sharing mosi, miso and sck is always normal situation. ;)

I am talking about situation when you have LCD, sd card, tmc2130 with spi, thermocouple max6675 with due board. If It should work with your adjustments can you share proper settings and cable connections? Shield on due does not matter for me.

tig33r, 2018-08-18:
No. I have LCD with sd card hw spi (standard configuration so I am assuming this is hw spi) and tmc sw spi (different pins than LCD with sd). Now I want to add max6675, could be anywhere I just want working solutions.

JackDaFlow, 2018-08-18:
Hey, Are you sure, that the SDCard is on HW_SPI? HW_SPI on DUE is in the center of the board, near the IC. Do you use a modified RAMPS? Here HW SPI is not located on the AUX 4, or any other PIN.

tig33r, 2018-08-18:
I have ramps-fd v2.1A and sdcard is connected to 3x2 goldpin more or less on due board center.  Board schematic https://github.com/bobc/bobc_hardware/blob/master/RAMPS-FD/RAMPS-FD-Schematic.pdf and adapter schematic https://reprap.org/mediawiki/images/3/36/LCD-panel-adapter-FD-Schematic.jpg

closed 2019-02-19 by thinkyhead