MarlinFirmware/Marlin #11577: [bugfix-2.0.x] Layer shift during print with TMC2130

dmitry-kutergin, 2018-08-17:
### Description

Layers shifts with the latest bugfix-2.0.x.

### Steps to Reproduce

1. Start printing a part,
2. After several hours I can see a layer shift 

**Expected behavior:**  I should print without layer shifts.

**Actual behavior:** Layer shift happens, sometimes multiple, mostly in Y direction, but sometimes in X,Y as well.

#### Additional Information
I use TMC2130 for my X,Y,Z,E0,E1. There was a similar issue with a huge investigation (can not find it now) that resulted in conclusion that the latest bugfix-2.0.x should fix it, apparently it is not. Anyways here are my configs attached. 
I tried lowering the speeds, accelerations and jerks, does not seem to help. With 1.1.8 it seemed to work fine with higher speed settings, when I last printed beginning of this year.
It usually happens after 6-7 hours of print, and I ruined a dozen of prints already.

[image]

[Configuration.h.gz](https://github.com/MarlinFirmware/Marlin/files/2298443/Configuration.h.gz)
[Configuration_adv.h.gz](https://github.com/MarlinFirmware/Marlin/files/2298444/Configuration_adv.h.gz)

dmitry-kutergin, 2018-08-17:
Now trying to print just the top part, since it happens more frequently closer to the top of the print, no issues so far.

GhostlyCrowd, 2018-08-17:
I've got the same experience and same driver setup. I've been pulling my hair out.  1 out of ever 3 prints of the same gcode will layer shift on the Y.

I've tried everything mechanical even replacing the stepper motor, belt and the driver. Making sure vref is good and giving it more RMS. And actively cooling the stepper even though it does not get over 45C

GhostlyCrowd, 2018-08-18:
Just lost an otherwise beautiful print as i was typing the above to this.

[image]

smoki3, 2018-08-18:
I also get some layer shifts with the latest build... 

CoreXY,
Trinamic 2130

larsnordstrom, 2018-08-19:
Layershifting can occur for several reason and to pinpoint the exact reason is pretty complex. To begin with:
- Do the drivers overheat.
- Do the motors get enough current.
- Acceleration is too high.
- Slipping belts.
- Too tight belts.
- Homing sensitivity too low.
- Triggering endstops.
- Someone bumped into the printer. (The cat...)
- Hybridthresholds are too low.
- Printing faster that what your machine can handle.
- Vibrations.
- Binding on the bearings.

These are the things I have found affect layershifting in the past year using tmc2130.

dmitry-kutergin, 2018-08-19:
Drivers do show temperature prewarn for x and y. I have active cooling and my current is set to just 900 ma. However prewarn was present with old version as well and it did not have the issue. Anyway making sure motors have enough current and drivers are not overheating are two contradicting points in my case. I will try to raise sensorless homing threshold. Why do you say it is related to hybrid thresholds, how they can trigger layer shifts? I have threshold set to 60mm/s btw. All other points seems to be irrelevant for my case.

larsnordstrom, 2018-08-19:
I found 600-700 to be enough for X/Y, 900 on Z and 450 on E. 

I run my hybrid at 130 on X/Y, Z at 10, E at 40. 

I’ve found that when thresholds are too low the machine can sometimes run just on the point where it is just about to move over to hybrid. In that case it can act as it does not have enough power for the speed it needs and therefore cause layershifting.

dmitry-kutergin, 2018-08-19:
So essentially you a running in StealthChop all the time? I actually came from StealthChop-only trying to solve the issue. When I set my current limits on 700 for X, Y I can not get even a single layer without skipped steps/layer shifts (my frame is CR-10 mechanically, with 3d printed belt tensioners, and 1.3A stock CR-10 steppers on Y,Z, and cheap 1.7A stepper on X).

GhostlyCrowd, 2018-08-19:
Being in stealthchop all the time makes this issue worse for me. 

I can safely say its not a mechanical issue for my self Ive even went as far as buying new steppers to try and resolve its. It appears to be a marlin bug with lin adv and stalthchop some say? 

I'm going to convert my tmc2130's to run on 24v to see if that solve it as some suggest it does.

dmitry-kutergin, 2018-08-20:
Yes I am waiting for a new steppers as well, also will try to disable "s curve acceleration" and "multi axis interpolation". Also will try with the 1.1.18 version I have from Dec 2017.

GhostlyCrowd, 2018-08-20:
S curve enabled or disabled ha not made a difference to me.

truglodite, 2018-08-20:
I solved layer shift by increasing the stallguard threshold to 8 (defaults, I started much lower), and setting current to 550mA. This is on an a8 with stock motors. I was having layer shift at 500mA and below; the lower the current the more often it shifted. So I'd say try upping current and stall thresholds within thermal limits.

GhostlyCrowd, 2018-08-20:
Thanks @truglodite  I will investigate this change as i had another print fail after 6 hours whilst only in speedcycle mode. SO its not shifting as quickly but still an issue. I have not had a successful print since i flashed the most recent pull from git 3 days ago :/ 

Before this i was on a pull from March. it had issues but at least i could get some prints done.  I might flash back as I really REALLY need to get some parts finished for my new printer I'm building from scratch.

GhostlyCrowd, 2018-08-21:
I'll add i have 1.7A rated steppers now and am giving them 900mA, which is Irms 1.25A i believe so i doubt current is an issue for me.

cfelicio, 2018-08-22:
Oddly enough, I'm also having these weird layer shifts that happen after a few hours, but I'm using A4988.

I got it to happen pretty consistently on a huge D20 I'm trying to print, let me try to flash back to an older release and see if it still happens (so far I was suspecting mechanical or electrical issues with the printer itself).

dmitry-kutergin, 2018-08-23:
Upping currents is not really an option for me, since I am already using 900 mAh, and it already gives me overheat prewarns on TMC2130.

GhostlyCrowd, 2018-08-23:
Flashing back has worked for me mostly. The print that would layer shift every time now has printed fine for me flashing back to the compile from a march clone of the 2.0 bugfix. 

after i get the parts i need done ill flash back to a current build and do some testing, also going to upgrade the steppers to 24V VMOT.

cfelicio, 2018-08-23:
does the old flash have junction deviation / s curve enabled?

GhostlyCrowd, 2018-08-23:
For me no, But nor did the latest flash either. I tried with it on and off.

GhostlyCrowd, 2018-08-27:
Alright so after 6 Successful prints in a row after Giving my TMC2130's 24V+ (Now at 35V) I am comfortable saying my random layer shifting issue is finally solved. I am on a current Marlin pull and have hybrid mode enabled and my speeds are 120mm/s travel and 80mm/s print. no shifting at all.

thinkyhead, 2018-09-04:
Finding a balance between adequate current and adequate cooling is quite tricky with TMC2130 drivers. It's especially tough with stepstick-based TMC2130s (as opposed to board-integrated like the Einsy RAMBo), as they don't heatsink very well, and sometimes they're sinked improperly. So having a good airflow over the drivers is essential.

GhostlyCrowd, 2018-09-04:
If i had to make a guess it was a combo of stepper heat and torque loss on 12v. less current and more torque on 35V, so less heat naturally. Solved my issues at the very least.  My drivers have always been actively cooled and as far as i know I've never triggered a thermal warning on them.  :)

dmitry-kutergin, 2018-09-08:
Ok, I think I also overcame the issue myself. I did not raise the voltage (I am running 14V). What I did is disabled S curves, and X,Y interpolation (not sure it matters), and what is more important I changed all my steppers to be at least 1.4 Amps, furthermore I changed my Y motor from 40mm to 60mm motor. Afterwards I set 900mAh current limit for X,Y,Z (was 900,900,800). Although I still see "OT prewarn was triggered" in M122. I did have 3 successful 10 hours prints in a row (which never happened before).

dmitry-kutergin, 2018-09-10:
I tried to raise print speed and jerk a bit (to the values i had on 1.18) and I got a layer shift again.
There were actually multiple layer shifts in a row, all of those appear to be happening on complete layers (after a Z movement to a new layer, not in the middle of the layer).

truglodite, 2018-09-10:
Fwiw, I forgot to mention I am running my 2130's at 20V. That allowed me to drop my current from 700 to 550, and I never get thermal flags (good heatsink and fan setup).

wtarantul, 2018-11-04:
I have the same problem on TEVO Tornado 2.0 (the most relevant for today). With version 1.1.9 there is no such problem. I'm trying to fix the problem for 3 days. There is a solution ?
[image]
[image]
[image]

dmitry-kutergin, 2018-11-07:
My solution is to move to 1.1.8 version. My suspicion is that the issue is somehow related to the tmc2130 library. I noticed layer shifts on 1.1.8 with tmc2130 with versions 2.2+, but it is stable with 2.0.1. I wanted to check the sources of the lib to see what changed there (the behavior of the steppers is definitely different between those two versions, especially at startup), but I did not have time yet. 
Another possible solution is to add more torque to the motors by switching to 24v.

italocjs, 2018-11-16:
I'm having issues with the tmc2130 too, losing steps randomly, i've got three heatsinks + cooling directly above the driver, so no overheating.  

but i'm running them on the standalone mode, could the firmware version interfere with it?

dammitcoetzee, 2018-11-18:
Do you have the right rsense value? I realized today that I had .10 instead of .22. Fixing this seemed to eliminate this problem.

Roxy-3D, 2018-11-18:
The Layer Shift is real.   And it isn't limited to TMC2130 chips.   We are investigating over in #12403.  It looks like 1.1.9 is OK.   But there are a number of people doing large prints with that just to make sure.  (Myself included)

Right now...   We are fairly confident the layer shift problem is in everything from Sept 21 to current in BugFix 2.0.0.   We are trying to figure out where the problem crept into the code base.

GhostlyCrowd, 2018-11-18:
Having stealthchop and or s curve accel enabled are both guaranteed repeatable layer shifts for me if that help. Upping my Vmot to 35v has alleviated my layer shifts otherwise.

dmitry-kutergin, 2018-11-19:
I experienced layer shifts on 2.0.0 end July beginning of August (so it is before Sep 21), before I switched to 1.1.8. The in one case i got a layer shift with 1.1.8 with tmc2130 Arduino lib of version 2.3.0, when I switched to 2.0.1 there were no layer shifts for at least 6 30 hours prints. So I really believe that it is related to the tmc2130 lib version, at least behavior of the steppers right after power up is different between 2.0.1 and 2.3.0. I did not have time myself to run through the diffs between those versions yet.
Also I am in the middle of upgrading to 24v mains, to boost print speed, so I will see if I can push my 1.1.8 even further without layer shifts.

Regarding rsense. How to understand what value I need to set if it should not be default one (0.1, 0.22 or something else)?

dammitcoetzee, 2018-11-21:
@dmitry-kutergin For me I discovered that I overlooked the entry on: https://reprap.org/wiki/EinsyRambo#MarlinFirmware

jeremykoerber, 2018-12-09:
> Having stealthchop and or s curve accel enabled are both guaranteed repeatable layer shifts for me if that help. Upping my Vmot to 35v has alleviated my layer shifts otherwise.

Hey all, just upgraded to an MKS Gen-L and 4xTMC2130s. Have heatsinks and active cooling, but getting Y layer shifting after several hours of printing. The PSU is 12V 30A. I have Y current set to 1200mA (been increasing it slowly which has helped but not solved the problem). Running in stealth chop only. I've heard running stealth chop on a 12V PSU won't give enough torque thus the layer shift. Lots of good ideas here that I will try, but I had a question specifically about "Upping my Vmot to 35v". Can someone elaborate on how to do that? I'm familiar with adjusting vref, and changing the output on the PSU, but I'm not fully understanding what would be involved with adjusting vmot voltage. Hopefully this isn't a stupid question. Pretty new to all of this. Thanks in advance.

GhostlyCrowd, 2018-12-09:
> Hey all, just upgraded to an MKS Gen-L and 4xTMC2130s. Have heatsinks and active cooling, but getting Y layer shifting after several hours of printing. Marlin is 1.1.9. The PSU is 12V 30A. I have Y current set to 1200mA (been increasing it slowly which has helped but not solved the problem). Running in stealth chop only. I've heard running stealth chop on a 12V PSU won't give enough torque thus the layer shift. Lots of good ideas here that I will try, but I had a question specifically about "Upping my Vmot to 35v". Can someone elaborate on how to do that? I'm familiar with adjusting vref, and changing the output on the PSU, but I'm not fully understanding what would be involved with adjusting vmot voltage. Hopefully this isn't a stupid question. Pretty new to all of this. Thanks in advance.

Basically i desoldered the vmot and gnd pin beside vmot and have it running on a buck converter that steps up my 12v to 35v at 6A.  this allws me to drive the steppers at 35v whilst not having to convery the rest of my 12v system.

italocjs, 2018-12-10:
> Basically i desoldered the vmot and gnd pin beside vmot and have it running on a buck converter that steps up my 12v to 35v at 6A. this allws me to drive the steppers at 35v whilst not having to [...]

nice idea!, i'll try doing this on my tmc2208,  
I managed to stop losing steps by using an old 2.0 firmware i had here, but i still needed to reduce the speed to 100mms while using full stealthchop because of the decreased torque.

jeremykoerber, 2018-12-10:
@GhostlyCrowd What buck converter are you using? I intend to do the same thing. So many to choose from. And do you feel like 35V at 6A will really make a difference over say 24V at 5A? I'm looking at this one: https://amzn.to/2SDF7Ky . Thanks!

GhostlyCrowd, 2018-12-10:
> @GhostlyCrowd What buck converter are you using? I intend to do the same thing. So many to choose from. And do you feel like 35V at 6A will really make a difference over say 24V at 5A? I'm looking at this one: https://amzn.to/2SDF7Ky . Thanks!

This is the one I used, works great. https://www.amazon.ca/Converter-10-32V-12-35V-Charger-Module/dp/B079GQZFBQ/ref=sr_1_cc_3?s=aps&ie=UTF8&qid=1544475846&sr=1-3-catcorr&keywords=12+to+35v+6A+converter

jeremykoerber, 2018-12-10:
Aww man, they won’t ship that to me in the US.. :(

https://www.amazon.ca/Converter-10-32V-12-35V-Charger-Module/dp/B079GQZFBQ/ref=sr_1_cc_3?s=aps&ie=UTF8&qid=1544475846&sr=1-3-catcorr&keywords=12+to+35v+6A+converter

GhostlyCrowd, 2018-12-10:
Amazon.com im sure has the same one. Give it a look.

jeremykoerber, 2018-12-10:
Cool. Hopefully this one does the trick. http://a.co/d/1ZhaGS7
None seem to ship from the US, but this one should only take a week to get
here.

https://www.amazon.ca/Converter-10-32V-12-35V-Charger-Module/dp/B079GQZFBQ/ref=sr_1_cc_3?s=aps&ie=UTF8&qid=1544475846&sr=1-3-catcorr&keywords=12+to+35v+6A+converter
<https://github.com/MarlinFirmware/Marlin/issues/11577#issuecomment-445972603>>,

jeremykoerber, 2018-12-18:
Just following up. I've got my layer shifting issue under control it seems. I'm using this buck converter: http://a.co/d/1ZhaGS7. @GhostlyCrowd I found that there is no way to get 6A at 35V with a 150W converter. 6A is the max current, and when you dial up the voltage the current goes down. I've found the best compromise to be at 24V. But even with that I still had layer shifting in stealthchop. The real solution was reducing acceleration way down. In any case, spreadcycle is a lot quieter with 24V which is nice since it's still used for homing. Thanks for all the help.

drphil3d, 2018-12-24:
same issue here with marlin 2.0 :(

Scorcerer, 2018-12-30:
Had the same issue on RAMPS 1.4 with TMC2130 during that big investigation someone mentioned - for me it was softSPI causing this - probably too many wires caused noise levels to get so high, my TMCs interpreted those as step/dir commands... was OK with standard hardware SPI though - now I'm waiting for RAMPS 1.6+ to get this on-board to compare.

dmitry-kutergin, 2019-01-03:
I actually finished converting my set-up to 24v. And I got weird results, now i have more layer shifts that I had before, with the same print settings. That sounds counter intuitive. I converted everything back to 12V yesterday, will see if it eliminates layer shifts as it was before 24v-conversion.

jeremykoerber, 2019-01-04:
@dmitry-kutergin After I switched to 24v I was also surprised to see that I still had layer shifts. I was able to resolve this by drastically lowering my acceleration settings across the board, and also reducing my jerk settings. Here are some helpful resources on this:

https://www.youtube.com/watch?v=c3v9E1AwDBE
https://www.youmagine.com/designs/multi-angle-test-tower-for-calibrating-jerk

I have my jerk at 15 and acceleration at 100 mm/s. Although I just realized that my X gantry on my CR-10 (dual Z steppers) wasn't completely level/tensioned properly. I hopefully just corrected that, and I'll see if that helps. My theory is that the print head was sometimes crashing into the print because of this, which caused step loss and layer shifting.

The other thing to try would be to disable stealthchop and only use spreadcycle for a test print (one where you previously had layer shifting). If you still have layer shifting in spreadcycle, your issue may be more mechanical/structural than electronic. But if it works, you probably just need to tune your acceleration and jerk. Hope this helps.

drphil3d, 2019-01-04:
When your in stealth chop you have to greatly lower acceleration I don’t
run above 500. Hybrid threshold can also cause issues of the change in
speed is too fast.
I also found that junction deviation is more likely to cause lost steps.

> --

Philip Mally
3D Printing & Rapid Prototyping Services
530-205-3210
pmally@gmail.com

dmitry-kutergin, 2019-01-04:
Ok, so i did a test print with 12v (14.4v actually), and I do not see layer shifts anymore. I only changed PS for the printer, no other changes. 
I understand that lowering speed/jerk/acceleration will help, but the whole idea of converting my set-up to 24v to be able to print faster, so lowering speed is not a solution for me. Currently I can safely print 800 grams prints (weight matters since Y axis need to drag it back and forth), with 100 mm/s perimeters, 140 mm/s infill, with 20/20/2 X/Y/Z jerk, and 1250/800/80 X/Y/Z acceleration on 12v. With the very same settings on 24v I have layer shifts for both axis. All is done in stealthchop, I am not using spreadcylce except for homing, because my printer stays in my living room and spreadcycle is too loud for overnight runs.
And my current limits are 900 mA for all axis. When I switched to 24v, I bumped x and Y to 1000 mA, which eliminated shifts in X axis, but Y axis was still shifting randomly. I can not bump my current more since even with 900 mA I have over-temperature pre-warns (and yes I am using heatsinks and active cooling on my tmc2130).

drphil3d, 2019-01-05:
You can’t run the stepper drivers over 960ma
Are you using hybrid threshold?
Just a guess but at 24 volts resistance and current changes with the
characteristics of the motors. I would assume the motors would run at lower
amperage at higher voltage.
You should start at a lower acceleration setting and work your way up. I
noticed it gets more likely to happen the higher up the layer height.
Add a cooling fan to your board and heat sinks to the stepper drivers, that
will help a lot.
-- 

Philip Mally
3D Printing & Rapid Prototyping Services
530-205-3210
pmally@gmail.com

italocjs, 2019-01-05:
> You can’t run the stepper drivers over 960ma Are you using hybrid threshold? Just a guess but at 24 volts resistance and current changes with the characteristics of the motors. I would assume the [...]

Things get a bit trickier here!, the driver is a constant current source, so *ideally* doesnt matter how much voltage do you have in the input, the driver will do its magic and provide enough to the stepper have the desired current.   The problem is, motors are coils, and coils have indutance.   When you try to move the stepper faster you significantly increase the inductance (resistance), thus, the default 12v might not be enough to provide the desired current. when this happens, the stepper will lose steps.  thats why its common to see faster cnc's (2000mms) using 48v, 96v or even more,   for 3dprinters, 24-30v is a good point to start when trying to work at high speeds (which probably wont go above 400mms).    I was having issues with a tmc at 12v and had to decrease its acceleration, it did the trick, but made my printings 30% slower.  so i used a step-up converter to change my 12v to 35v, and connected it only to the VMOT pin in the driver (disconnecting it from the ramps), now i'm able to use my tmc again with 4000 acceleration, the printers is even more silent.

italocjs, 2019-01-05:
> 
> 
> Ok, so i did a test print with 12v (14.4v actually), and I do not see layer shifts anymore. I only changed PS for the printer, no other changes.
> I understand that lowering speed/jerk/acceleration will help, but the whole idea of converting my set-up to 24v to be able to print faster, so lowering speed is not a solution for me. Currently I can safely print 800 grams prints (weight matters since Y axis need to drag it back and forth), with 100 mm/s perimeters, 140 mm/s infill, with 20/20/2 X/Y/Z jerk, and 1250/800/80 X/Y/Z acceleration on 12v. With the very same settings on 24v I have layer shifts for both axis. All is done in stealthchop, I am not using spreadcylce except for homing, because my printer stays in my living room and spreadcycle is too loud for overnight runs.
> And my current limits are 900 mA for all axis. When I switched to 24v, I bumped x and Y to 1000 mA, which eliminated shifts in X axis, but Y axis was still shifting randomly. I can not bump my current more since even with 900 mA I have over-temperature pre-warns (and yes I am using heatsinks and active cooling on my tmc2130).
> PS. I am using 1.1.8 and tm2130 library of 2.0.2 version, so there is no fancy options like junction deviation or s-curve acceleration there yet.

you might want to try adding a fan pointing to the tmc's and more heatsinks to your drivers, this seems like overheating (those tmcs are pretty badly designed, the chip upside down doesnt allow for a good cooling).  i'm using three heatsinks (one above the other) from a4988 chips a had laying around, they helped a lot!

dmitry-kutergin, 2019-01-06:
Thanks for the comments.

> You can’t run the stepper drivers over 960ma
Are you using hybrid threshold?
Just a guess but at 24 volts resistance and current changes with the
characteristics of the motors. I would assume the motors would run at lower
amperage at higher voltage.

No as I stated I am using only stealthchop, I actually tried to switch to spread cycle all the way, and it did not give any improvements in print speed. How do you know about 960 mA limit? I read several times people were running TMC drivers on over 1 A currents. And yes the whole point in current limiting driver is to be able to provide constant current no matter what voltage you are supplying to the motor.

> When you try to move the stepper faster you significantly increase the inductance (resistance), thus, the default 12v might not be enough to provide the desired current. when this happens, the stepper will lose steps. thats why its common to see faster cnc's (2000mms) using 48v, 96v or even more

Yes I read about it as well, even with graphs showing how much more high speed torque steppers are having whe bumping from 12v to 24/36v. This point was the main reason I started this 24v conversion, but the results I got are kind of contradicting with this theory, I am confused here. I guess bumping to 35v might bring me to the where I am right now with my 14v, but what is the point?

> you might want to try adding a fan pointing to the tmc's and more heatsinks to your drivers, this seems like overheating (those tmcs are pretty badly designed, the chip upside down doesnt allow for a good cooling). 

I have heatsinks on all 5 tmc driver I have, and I have two 60mm fans and one 50mm fan blowing on them, not sure how can i improve my cooling even more.

My thinking was maybe I need higher resistance motors to work effectively on 24v? Cos mine are 1.3-1.6 Amps. Maybe I need 0.4-0.6 A? They are rated at ~12v so at 24v they can run 900 mA currents i have on my drivers. It would be 2x overload to them though. 
Anyway I guess I am stuck with my current printer frame and motors at 100-140 mm/s.

larsnordstrom, 2019-01-06:
Here's what I am running on 2.0 without any layershifting with 1A motors and an Einsy Rambo board. The only issue I have is grinding sound from extruder when using LIN_ADVANCE, however I have that deactivated in my startscript.

DEFAULT_MAX_FEEDRATE          { 200, 200, 12, 120 }
DEFAULT_MAX_ACCELERATION      { 1000, 1000, 200, 5000 }
DEFAULT_ACCELERATION                   1250 
DEFAULT_RETRACT_ACCELERATION  1250
DEFAULT_TRAVEL_ACCELERATION      1250
S_CURVE_ACCELERATION

JUNCTION_DEVIATION
ADAPTIVE_STEP_SMOOTHING
R_SENSE     0.22
HOLD_MULTIPLIER    0.5
INTERPOLATE       true 
X_CURRENT          636 
X_MICROSTEPS       16 
Y_CURRENT          636
Y_MICROSTEPS       16
Z_CURRENT          900
Z_MICROSTEPS       16
E0_CURRENT         636
E0_MICROSTEPS       32
CHOPPER_TIMING CHOPPER_DEFAULT_24V
X_STALL_SENSITIVITY  5
Y_STALL_SENSITIVITY  8
Z_STALL_SENSITIVITY  6

boelle, 2019-02-20:
@dmitry-kutergin still have problems?

dmitry-kutergin, 2019-02-23:
I do not know about 2.0.x branch since I did not tried it for 6 months. But 1.1.8 works fine for me.
I found a sweet spot with print/movement speeds no more than 100 mm/s per axis, but I bumped accelerations to 2000 mm/s^2. It prints even faster than with 140 mm/s print speeds and 1000 mm/s^2 accel. The only downside with higher acceleration is shaking, that sometime knocks off my IR heaters.

boelle, 2019-02-23:
printing faster is rarely better, print quality goes down, printer can shake itself to 1000 bits

but you should give 2.0 a test when you can, its here things happen these days and its both 32 bit and 8 bit

dmitry-kutergin, 2019-02-23:
yes if you have small parts you can have a choice to print slow. But when your part takes days to complete, and you are running all that time 600w worth of heaters to prevent warping, it is reasonable to sacrifice quality a bit, as long as it is only exterior quality, and layer bonding does not suffer, and make it print faster.

dmitry-kutergin, 2019-02-23:
As for 2.0, I would like to try but last time I spend 1.5 months and did not get a single completed print, so as long as it works for me with 1.1.8 until I finish my project I can not afford experiments.

boelle, 2019-02-23:
i never compromise on quality, things just take the time they take,

i have some upcoming prints that take 40 hours and i dont speed it up :-D

boelle, 2019-02-23:
at least you should edit title to reflext you are using 1.1.8 and not 2.0, just so we dont confuse the dev people :-D

[23 more replies]

closed 2019-07-11 by thinkyhead