MarlinFirmware/Marlin #11582: [1.1.9] Very strange G28 behavior (almost hits Z limit going up)

CostinGhiocel, 2018-08-18:
### Description

Printer: Creality CR-20 (new one)

When performing a G28 Home all, the printer will perform both X and Y touches twice, going up between them. Initially, it goes up in small increments. If performed after a print (any print, including printing the UBL validation) it will go up significantly higher, almost like it's stuck in relative mode. It almost maxes out Z if i'm not careful to bring the x carriage down.

I initially was running 1.1.8 without this issue present, i updated to 1.1.9 and it started since then.

### Steps to Reproduce

1) Finish or cancel a print
2) Perform G28

#### Additional Information

Config files:

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/2299150/Marlin.zip)

Video showing the behavior:

https://youtu.be/hIihjOswZXM

smoki3, 2018-08-18:
have you tried the latest version? It should be fixed with the latest commit.

CostinGhiocel, 2018-08-18:
Oh just saw the commit from 2 days ago, I thought they killed updates on this one. Thanks for the heads up, I'll give it a shot

closed 2018-08-19 by CostinGhiocel