MarlinFirmware/Marlin #11585: [2.0.x] Can't compile with BOARD_ULTIMAKER selected

KlausHans, 2018-08-18:
### Description

Hello, i use a GT2560 Board and the manufacturer (Geetech) used BOARD_ULTIMAKER as definition, so did i on 1.1.9. Now i wanted to go to 2.0 and unfortunately a compile error appears:

```
In file included from sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/HAL.h:38:0,
                 from sketch/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:28,
                 from sketch/src/HAL/HAL_AVR/MarlinSerial.cpp:38:
sketch/src/HAL/HAL_AVR/../../inc/../pins/pins_ULTIMAKER.h:112:33: error: pasting "/* PIN that has to be turned on right after start, to keep power flowing.*/" and "_DDR" does not give a valid preprocessing token
 #define SUICIDE_PIN        54   // PIN that has to be turned on right after start, to keep power flowing.
                                 ^
sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/fastio_AVR.h:84:42: note: in definition of macro '_SET_OUTPUT'
 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
                                          ^
sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/fastio_AVR.h:102:35: note: in expansion of macro 'SET_OUTPUT'
 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                   ^
sketch/src/HAL/HAL_AVR/../../Marlin.h:224:27: note: in expansion of macro 'OUT_WRITE'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                           ^
sketch/src/HAL/HAL_AVR/../../Marlin.h:224:37: note: in expansion of macro 'SUICIDE_PIN'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                                     ^
sketch/src/HAL/HAL_AVR/../../inc/../pins/pins_ULTIMAKER.h:112:33: error: pasting "/* PIN that has to be turned on right after start, to keep power flowing.*/" and "_PIN" does not give a valid preprocessing token
 #define SUICIDE_PIN        54   // PIN that has to be turned on right after start, to keep power flowing.
                                 ^
sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/fastio_AVR.h:84:61: note: in definition of macro '_SET_OUTPUT'
 #define _SET_OUTPUT(IO)       SBI(DIO ## IO ## _DDR, DIO ## IO ## _PIN)
                                                             ^
sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/fastio_AVR.h:102:35: note: in expansion of macro 'SET_OUTPUT'
 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                   ^
sketch/src/HAL/HAL_AVR/../../Marlin.h:224:27: note: in expansion of macro 'OUT_WRITE'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                           ^
sketch/src/HAL/HAL_AVR/../../Marlin.h:224:37: note: in expansion of macro 'SUICIDE_PIN'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                                     ^
In file included from sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/HAL.h:38:0,
                 from sketch/src/HAL/HAL_AVR/../../inc/MarlinConfig.h:28,
                 from sketch/src/HAL/HAL_AVR/MarlinSerial.cpp:38:
sketch/src/HAL/HAL_AVR/../../inc/../pins/pins_ULTIMAKER.h:112:33: error: pasting "/* PIN that has to be turned on right after start, to keep power flowing.*/" and "_RPORT" does not give a valid preprocessing token
 #define SUICIDE_PIN        54   // PIN that has to be turned on right after start, to keep power flowing.
                                 ^
sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/fastio_AVR.h:79:48: note: in definition of macro '_WRITE'
 #define _WRITE(IO,V)          do{ if (&(DIO ## IO ## _RPORT) < (uint8_t*)0x100) _WRITE_NC(IO,V); else _WRITE_C(IO,V); }while(0)
                                                ^
sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/fastio_AVR.h:102:51: note: in expansion of macro 'WRITE'
 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                                   ^
sketch/src/HAL/HAL_AVR/../../Marlin.h:224:27: note: in expansion of macro 'OUT_WRITE'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                           ^
sketch/src/HAL/HAL_AVR/../../Marlin.h:224:37: note: in expansion of macro 'SUICIDE_PIN'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                                     ^
sketch/src/HAL/HAL_AVR/../../inc/../pins/pins_ULTIMAKER.h:112:33: error: pasting "/* PIN that has to be turned on right after start, to keep power flowing.*/" and "_WPORT" does not give a valid preprocessing token
 #define SUICIDE_PIN        54   // PIN that has to be turned on right after start, to keep power flowing.
                                 ^
sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/fastio_AVR.h:69:21: note: in definition of macro '_WRITE_NC'
   if (V) SBI(DIO ## IO ## _WPORT, DIO ## IO ## _PIN); \
                     ^
sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/fastio_AVR.h:91:31: note: in expansion of macro '_WRITE'
 #define WRITE(IO,V)           _WRITE(IO,V)
                               ^
sketch/src/HAL/HAL_AVR/../../inc/../HAL/HAL_AVR/fastio_AVR.h:102:51: note: in expansion of macro 'WRITE'
 #define OUT_WRITE(IO,V)       do{ SET_OUTPUT(IO); WRITE(IO,V); }while(0)
                                                   ^
sketch/src/HAL/HAL_AVR/../../Marlin.h:224:27: note: in expansion of macro 'OUT_WRITE'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
[... 106 lines cut ...]
sketch/src/HAL/HAL_AVR/../../Marlin.h:224:27: note: in expansion of macro 'OUT_WRITE'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                           ^
sketch/src/HAL/HAL_AVR/../../Marlin.h:224:37: note: in expansion of macro 'SUICIDE_PIN'
   inline void suicide() { OUT_WRITE(SUICIDE_PIN, LOW); }
                                     ^
exit status 1
Fehler beim Kompilieren für das Board Arduino/Genuino Mega or Mega 2560.

```

### Steps to Reproduce

1. Download Marlin Bugfix 2.0 Branch
2. Download and install Arduino 1.9 Beta
3. In configuration.h set BOARD_ULTIMAKER in line 136
4. Try to compile

**Expected behavior:** It compiles.

**Actual behavior:** It gives the compile error, see above.

#### Additional Information

I don't know if it matters, but i use Ubuntu 18.04.

p3p, 2018-08-18:
This is a bug in the Arduino (possibly avr_gcc) preprocessor, the macro expansion is including comments that are on the same line, you could create an issue on their repository. To work around the problem you would need to remove the comments from the pin definitions lines in pins_ULTIMAKER.h.

KlausHans, 2018-08-18:
You are right, i deleted some comments and it worked. Strange thought, 
`#define TEMP_0_PIN          8   // Analog Input`
(and other analog input comments) made no problem.

closed 2018-08-18 by KlausHans

per1234, 2018-08-24:
I believe this is the relevant bug report on the Arduino side:
https://github.com/arduino/arduino-preprocessor/issues/6
This was introduced with the new arduino-preprocessor tool in Arduino IDE 1.8.6. It's really unfortunate that Arduino did a production release of the IDE using a tool with multiple serious bugs that were reported long ago.

per1234, 2018-08-26:
It looks like I was wrong about this being arduino/arduino-preprocessor#6. See https://github.com/arduino/Arduino/issues/7930#issuecomment-416079657.