MarlinFirmware/Marlin #11590: [bugfix-2.0.x]  Temperature readings unstable

hamster65, 2018-08-19:
[image]
This is a picture taken about 2 weeks ago. Bed and nozzle temperature gave unstable readings. It stopped as soon as the steppers were activated (for example at the 2 Min mark) and it started once I deactivated them.

[image]
This is taken today. Both prints failed with a thermal runaway (nozzle). 
This happens after the first layer, when the nozzle temperature is reduced and, at the same time, the layer fan switches on. However, this was never a problem before. 
There seems to be still noise on the extruder temperature reading and at the left, it also seems unstable for the bed sensor. The dips at 17 Min and 7 Min can be ignored (g-code related). I could not find a connection to the stepper state.

Also, sometimes the printer hangs on "E0 heating" forever, and the temperature reading in the lcd is jumping arround. I checked all connections and my ground paths and did not find issues here.
Is anyone observing temperature fluctuations as well?

I'm also wondering if there is some kind of hysteresis for Gcodes like M109 (wait for Hotend temperature). Or does it wait until it stabilizes to the exact temperature? I'd think +- 1°C would be a sufficient margin.

kenan-kajkus, 2018-08-19:
Did you run PID-Tuning? Looks like not proper PID-values

hamster65, 2018-08-19:
Yes, but not with the layer fan running because it gets switched of during tuning. 
https://github.com/MarlinFirmware/Marlin/issues/11589

Kp=52,18
Ki=8,97
Kd=75,87

comps, 2018-08-20:
Maybe related to https://github.com/MarlinFirmware/Marlin/issues/11323 .

thinkyhead, 2018-08-21:
> I'm also wondering if there is some kind of hysteresis for Gcodes like M109 

```cpp
// Extruder temperature must be close to target for this long before M109 returns success
#define TEMP_RESIDENCY_TIME 10  // (seconds)
#define TEMP_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.

// Bed temperature must be close to target for this long before M190 returns success
#define TEMP_BED_RESIDENCY_TIME 10  // (seconds)
#define TEMP_BED_HYSTERESIS 3       // (degC) range of +/- temperatures considered "close" to the target one
#define TEMP_BED_WINDOW     1       // (degC) Window around target to start the residency timer x degC early.
```

hamster65, 2018-08-23:
Thanks thinkyhead! I'll increase the margin. Just today, my printer just waited forever while temperatures were jumping around.
As for the thermal runaway, I think it is a different problem which has nothing to do with the unstable readings.

hamster65, 2018-08-26:
I swapped my E3D Hotend to a volcano version and ran PID autotune (M303 S210 U C10).
 This is what happens while printing. 
[image]
PID values: 
Kp 35.90
Ki 5.22
Kd 61.75

hamster65, 2018-08-28:
I changed one cable to 2,5mm². Everything except for the heated bed runs through this fairly long cable. With the new cable, the leds stopped flickering when the nozzle is heating up. 

However, it did not stop the temperature issue and I cannot explain the readings:
[image]
From 13:24 to 13:31, only the heated bed was active. Its running in bang bang mode, so most of the time, it did not draw current. Fans, steppers were disabled. So even when there is almost no current flow, fluctuations occur for both sensors.
When the nozzle is heated to 50°C, this the nozzle sensor does not show the fluctuations any more, which again is strange.

Power supply for the arduino mega looks clean.
[image]

hamster65, 2018-09-02:
Turned out to be a bad crimp contact on the connector to the ramps board.

closed 2018-09-02 by hamster65