MarlinFirmware/Marlin #11591: [bugfix-2.0.x] Homing / BLTouch not deployed , Nozzle crash in bed glass

Stephane-80, 2018-08-19:
Hello Y have upgraded to the today version 
and now  BLTouch never gets deployed during homing or by the Bltouch menu in marlin
someone else has this problem?
before this upgrade with the same config it worked
thanks , 
best regards , Stephane

Stephane-80, 2018-08-19:
Maybe I have a hardware problem , sorry

julianCast, 2018-09-07:
My bltouch will deploy but it wont stop when contact the bed and it will crash my nozzle. So tired of this.

julianCast, 2018-09-08:
This is for those with the same problem:
I fixed it ! I can't believe it was just the height (the distance between nozzle and bltouch), I trusted a downloaded mount, I modified just 1 mm and it work perfect everytime. Hardware problems are the best. phew.

boelle, 2019-02-20:
@Stephane-80 problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead