MarlinFirmware/Marlin #11593: Autotune: Get always PID suggestions like ~5/~0.2/~17 which are far off

dgrat, 2018-08-19:
Working PID:
Last days I tried to get some working PIDs on my printer. My setup seems to work relatively well with PID-Terms in the range of ~5/~1/~35-45 and probably even higher D-Terms, Leading to 1°C oscilations.  

Autotune:
In the last days I repeated Autotune over and over, with and without fan and for a temperatures of 240 and 280°C. I alwas get PIDs in the same range of ~5/~0.2/~17 Plus minus ~0.5/~0.1/~2. The consistence would be awesome if the result would be working. The D-term is way to low in relation to the PI-terms. The I-term on the other hand is too low to keep up with the cooling fan. The temperatures with these PIDs oscilates during tuning by 10-15 degrees and during print by at least 5. 

Observation:
The tuning function seems to oscilate around these wrong PID-terms with mini steps forth and back, over and over. 

Conclusion:
Neither the PIDs nor the min/max temperature variation change . Autotune should fail at this point, but it continues for several runs :/

Suggestion:
Would it make sense to start with a fixed P/I-term and first optimize the D-term with bigger increment? Later one can increase the I-term and P. 

Question:
Are there maybe some settings in the config file to improve Autotune? The allowed min/max deviation of 10 degrees seems wrong at least in my file?!

RudyBenner, 2018-08-20:
You might try a little lower in 
#define BANG_MAX 225
instead of 255.
should settle in 2-3 oscillations.

dgrat, 2018-08-20:
Actually, I converted to a 24V system and lowered BANG_MAX to 48. It works till 72. Beyond this the overshoot are too high (or the D-term too small?).

closed 2019-02-19 by thinkyhead