MarlinFirmware/Marlin #11595: Marlin 1.19 and filament width sensor

stefi01, 2018-08-20:
I transfered all the settings from marlin 1.18 to 1.19 and i get an error with the filament width sensor.
works as it should without problem in 1.18 and the settings were double checked against 1.18

temperature.cpp:1067: error: 'FILWIDTH_ERROR_MARGIN' was not declared in this scope

 if (ABS(filament_width_nominal - filament_width_meas) <= FILWIDTH_ERROR_MARGIN)

                                                          ^
sketch\temperature.cpp: In static member function 'static void Temperature::readings_ready()':

temperature.cpp:1752: error: 'raw_filwidth_value' was not declared in this scope

 current_raw_filwidth = raw_filwidth_value >> 10;  // Divide to get to 0-16384 range since we used 1/128 IIR filter approach

I use these settings in 1.18 without a problem, i have no idea what   #define FILWIDTH_ERROR_MARGIN        1  is ment to be set to but that also works pretty well as shown in 1.18

[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/2300858/Configuration.txt)
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/2300859/Configuration_adv.txt)

gloomyandy, 2018-08-20:
It looks like there are extra settings in 1.1.9 so you can't just copy your 1.1.8 settings over (this is true in general, you need to merge your old settings with the current ones). The 1.1.9 sample config has the following settings in conviguration_adv:
```
if ENABLED(FILAMENT_WIDTH_SENSOR)
  #define FILAMENT_SENSOR_EXTRUDER_NUM 0    // Index of the extruder that has the filament sensor. :[0,1,2,3,4]
  #define MEASUREMENT_DELAY_CM        14    // (cm) The distance from the filament sensor to the melting chamber

  #define FILWIDTH_ERROR_MARGIN        1.0  // (mm) If a measurement differs too much from nominal width ignore it
  #define MAX_MEASUREMENT_DELAY       20    // (bytes) Buffer size for stored measurements (1 byte per cm). Must be larger than MEASUREMENT_DELAY_CM.

  #define DEFAULT_MEASURED_FILAMENT_DIA DEFAULT_NOMINAL_FILAMENT_DIA // Set measured to nominal initially

  // Display filament width on the LCD status line. Status messages will expire after 5 seconds.
  //#define FILAMENT_LCD_DISPLAY
#endif
```
As you can see this includes the missing FILWIDTH_ERROR_MARGIN. Do you still have problems with this value defined?

stefi01, 2018-08-20:
hi, when i say copied over i mean i typt in the settings in to the new 1.19
config h and adv but the above settings were copied from 1.18 although
those 4 lines of code are exactly the same in 1.19 including the error
margin line but yeah it still does not work, any thoughts would be
appreciated :)

ve3sjk, 2018-09-25:
I am also seeing this error in 1.1.9 has there been any fixes for this yet. My sensor is sending 1.80 volts on pin A5 and the display is showing zero if i add the declaration static unsigned long raw_filwidth_value = 0; to the filament width sensor settings in configuration_adv. I can see the declaration in temperature.cpp but it isn't being called during compile.

   #if ENABLED(FILAMENT_WIDTH_SENSOR)
    static unsigned long raw_filwidth_value = 0;
  #endif

ve3sjk, 2018-09-27:
I have not fully tested this yet after making the following changes to the Temperature.cpp file i was able to get past the compiler missing declaration error and can now see the filament diameter on my full graphics controller which matches the actual voltage on the A5 pin.

Please note i am using the 1.1.9 most recent release or at least what was most recent about 2 or 3 weeks ago so the line numbers might vary and my default configuration file was for Anet A8.

In temperature.cpp 

On lines 1894,1895,1896 within the Temperature::isr() routine you will find:

`
  #if ENABLED(FILAMENT_WIDTH_SENSOR)
    static unsigned long raw_filwidth_value = 0;
  #endif

` 

In my case i cut the lines out of that routine and pasted them
back into the file on lines 1746,1747,1748 which is located
just BEFORE 

the Temperature::readings_ready() subroutine like below

`
  #if ENABLED(FILAMENT_WIDTH_SENSOR)
    static unsigned long raw_filwidth_value = 0;
  #endif

  void Temperature::readings_ready() {

`

I get a reading of 1.84 at 100% when powering up and after issuing an M405 command the reading changes to 1.84 at 88% which appears that the sensor is working now. I still have to figure out the proper settings in Simplfy3d to use the sensor for a print which is why I am not saying this 100 percent fixes the issue.

ve3sjk, 2018-09-27:
Just an update. I have done a test print by setting filament diameter in Simplify3d to 1.75 and issued an M405 in the startup script. The sensor is adjusting the extrusion factor based in the diameter reading so it looks like the above fix worked. One issue i do see is that my oscilloscope says the A5 input is 1.78 volts were as the ramps/Arduino is reporting it on the lcd as 1.82 so it appears there is an offset of around 0.05 to 0.06 volts between the actual voltage on the pin and what A5 reads on the display. Not sure if i can compensate for this in the code. the print itself was a full bed on an Anet A8 using the filament width sensor and auto bed leveling it came out perfect.

AnHardt, 2018-09-27:
0.05-0.06V is at a 5V system about 0.1%. That's at a 10 bit ADC about 1 LSB. ->  Much better than expected.

ve3sjk, 2018-09-28:
You are correct , I also realized that the firmware is averaging the readings
In the subroutine so what I was seeing on my scope was just the rms
Value at the instance I looked at it. I have recalibrated the sensor with
The proper calibration rod and its down to 0.02 now which is very
good and my prints look great.

boelle, 2019-02-20:
@stefi01 problem solved?

stefi01, 2019-02-20:
hi, yeah it is now solved, was a bug in marlin and scot patched it some time ago

closed 2019-02-20 by stefi01