MarlinFirmware/Marlin #11598: Marlin 1.1.9 and skew correction

DaveAhrendt, 2018-08-20:
Marlin 1.1.9 with Folgertech i3 2020

I'm trying to understand how to implement skew correction.  If I understand the comments in configuration.h I can either enter my measurements for XY_DIAG_AC, etc OR define XY_SKEW_FACTOR.
It seems that XY_SKEW_FACTOR overrides my enter measurements but if I comment out  XY_SKEW_FACTOR I get a compile error.

#if ENABLED(SKEW_CORRECTION)
  // Input all length measurements here:
//  #define XY_DIAG_AC 282.8427124746
//  #define XY_DIAG_BD 282.8427124746
//  #define XY_SIDE_AD 200
  #define XY_DIAG_AC 141,1    //08/19/18
  #define XY_DIAG_BD 141.6
  #define XY_SIDE_AD 100.6

  // Or, set the default skew factors directly here
  // to override the above measurements:
  #define XY_SKEW_FACTOR 0.0

If I leave  XY_SKEW_FACTOR defined when I do M852 I get all zeros and the print doesn't change.
I can put some value in  XY_SKEW_FACTOR and it does get displayed via M852. Haven't tried another print.
I assume I'm doing something wrong but don't understand the comments.

Ccleorina87, 2018-08-20:
https://www.youtube.com/watch?v=YfAb5IaHDSo&t=1s

these could help you

DaveAhrendt, 2018-08-20:
Thanks for the link but it just shows that what I am doing is what I should be doing apart from needing to multiply the measurements by 2.  In any case I noted that he was using 1.1.8 so I tried that and it also fails to compile.  I then tried 1.1.8 with only the folger i3-2020 example config files and just enabled SKEW_CORRECTION and commented out the XY_SKEW_FACTOR define.  Here is the results of the compile.

Arduino: 1.8.6 Hourly Build 2017/11/07 04:33 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Dave\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Dave\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Dave\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10806 -build-path C:\Users\Dave\AppData\Local\Temp\arduino_build_994852 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\Dave\Google Drive\3D Printer\Folger 2020\Firmware\Marlin-Release\Marlin-1.1.8\Marlin-1.1.x\Marlin\Marlin.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Dave\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Dave\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Dave\Documents\Arduino\libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10806 -build-path C:\Users\Dave\AppData\Local\Temp\arduino_build_994852 -warnings=default -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.1.1 -prefs=runtime.tools.avrdude.path=C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9 -prefs=runtime.tools.avr-gcc.path=C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2 -verbose C:\Users\Dave\Google Drive\3D Printer\Folger 2020\Firmware\Marlin-Release\Marlin-1.1.8\Marlin-1.1.x\Marlin\Marlin.ino
Using board 'mega' from platform in folder: C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21
Using core 'arduino' from platform in folder: C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21
Detecting libraries used...
"C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\Dave\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Dave\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\mega" "C:\Users\Dave\AppData\Local\Temp\arduino_build_994852\sketch\Marlin.ino.cpp" -o "nul"
"C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\Dave\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Dave\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\mega" "-IC:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src" "C:\Users\Dave\AppData\Local\Temp\arduino_build_994852\sketch\Marlin.ino.cpp" -o "nul"
"C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\Dave\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Dave\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\variants\mega" "-IC:\Program Files (x86)\Arduino\libraries\LiquidCrystal\src" "C:\Users\Dave\AppData\Local\Temp\arduino_build_994852\sketch\G26_Mesh_Validation_Tool.cpp" -o "nul"
"C:\Users\Dave\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10806 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR   "-IC:\Users\Dave\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21\cores\arduino" "-IC:\Users\Dave [... cut ...]
C:\Users\Dave\AppData\Local\Temp\arduino_build_994852\sketch\Conditionals_post.h:856:46: note: in expansion of macro '_GET_SIDE'

     #define _SKEW_FACTOR(a,b,c) _SKEW_SIDE(a,_GET_SIDE(a,b,c),c)

                                              ^

C:\Users\Dave\AppData\Local\Temp\arduino_build_994852\sketch\Conditionals_post.h:861:11: note: in expansion of macro '_SKEW_FACTOR'

           _SKEW_FACTOR(XY_DIAG_AC, XY_DIAG_BD, XY_SIDE_AD)

           ^

Using library LiquidCrystal at version 1.0.7 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
'sqrt(2.91072e+5)' is not a constant expression

gloomyandy, 2018-08-20:
A couple of possible things in the line
```
#define XY_DIAG_AC 141,1 //08/19/18
```
You have a comma "," rather than a decimal point "." you might also want to try removing the comment as I think some versions of the Arduino compiler had issues with this.

DaveAhrendt, 2018-08-20:
Thanks gloomyandy for pointing out my boneheaded typo.  There is a big difference between a , and a .
I'll go away now and sulk in the corner.

closed 2018-08-21 by thinkyhead