MarlinFirmware/Marlin #11599: Documentation on how to tune Junction Deviation

Itox001, 2018-08-20:
Hello! Just updated to the new 2.0 bugfix and so far looking good, movement seems smoother with s curve acceleration, adaptive step smoothing and all the misc fixes and improvements done to the planner! 

One of the big features I wanted to try is junction deviation, but as soon as I was going to enable I noticed that I don't really know how to tune it. I've actually read through all discussions involving it here on the Marlin's git but practically I'm still not sure how to go about it.

From what I understood this are the key points:

*Max acceleration is used in the calculation (so default acceleration no longer has any effect?)
*The junction deviation value relates this Accel to the effects old jerk had

But how do I start? I guess I could use the defaults and go from there, but I'm not sure if I should lower Junction deviation value or acceleration if I find too much ringing, or even a starting point for my Max Accel. For example, I'm trying this on a tevo tornado, I use 500 printing Accel and 1000 travel Accel, 7 jerk for x and y. What values would be sensible to start with? How do I tune travel and printing Accel if there is only one single max Accel? I saw the formula relating max Accel and junction deviation value to old jerk, so I could do that to get a starting point with the default 0.02 junction deviation, but how do I go from there?

Thanks for the awesome work Marlin team, you're doing the world a great service!

mylife4aiurr, 2018-08-27:
I would always appreciate a how to tune junction deviation.... i leave it @ default because idk how to tune

marcio-ao, 2018-08-28:
I second this request :)

boelle, 2019-02-20:
i dont know where i stand on this one, but i have not figured how to tune it either

ArtemKuchin, 2019-02-22:
I kinda understand how to tube it. Basically there are two parameters: acceleration and deviation. The higher the acceleration the higher will be speed at corners, also, the higher the deviation also, the higher speed at corners. Deviation applies only for corners, acceleration for everything. So, set needed acceleration, print, check your corners, if okay then okay, if not, reduce either acceleration or deviation.

However, what i do not understand now is how it behaves on just streight lines.
Let the axis rest. 

How it was before.
ACC=1000; jerk=10. I say G1 X100 F6000
Right after that ut start with jerk speed of 10 mm/sec and the accelerates to set speed using ACC=1000mm/s*s

How does it work now? It start to accelerate to set speed using acc=1000 right from 0mm/s or
from MINIMUM_PLANNER_SPEED ?

joaomamede, 2019-07-31:
Still no clear guide? :)

The-slunk, 2020-01-29:
According to [This guide](http://blog.kyneticcnc.com/2018/10/computing-junction-deviation-for-marlin.html?m=1) its measured by .4*jerk*jerk/printing acceleration so for @Itox001 it would be .4*7*7/500 to give you a JD value of .04

ArtemKuchin, 2020-01-29:
Well, the question still remains:
How does it work now for streight lines when starting movement from rest. 
Does it start to accelerate to set speed using acc=1000 right from 0mm/s or
from MINIMUM_PLANNER_SPEED ?

qwewer0, 2021-01-01:
Could this be closed?
https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/Configuration.h#L806-L807

Sineos, 2021-01-01:
Uhmm, I would not exactly call a code comment a documentation 😕

qwewer0, 2021-01-01:
> Uhmm, I would not exactly call a code comment a documentation 😕

True, but there isn't too much to add to http://blog.kyneticcnc.com/2018/10/computing-junction-deviation-for-marlin.html?m=1

mydevpeeps, 2021-09-20:
I would like to suggest a code change for the junction deviation section to allow it to calculate. Here is an example:

`#define DEFAULT_EJERK    5.0  // May be used by Linear Advance

/**
 * Junction Deviation Factor
 *
 * See:
 *   https://reprap.org/forum/read.php?1,739819
 *   https://blog.kyneticcnc.com/2018/10/computing-junction-deviation-for-marlin.html
 */
#if DISABLED(CLASSIC_JERK)
  #if ENABLED(DEFAULT_ACCELERATION)
    #if ENABLED(DEFAULT_EJERK)
      #define JUNCTION_JERK DEFAULT_EJERK     // jerk value for junction deviation calculation (use already defined constant)
    #else
      #define JUNCTION_JERK 5.0               // jerk value for junction deviation calculation (manually define the jerk value for the formula if needed)
    #endif
    #define JUNCTION_DEVIATION_MM ( 0.4 * JUNCTION_JERK * JUNCTION_JERK / DEFAULT_ACCELERATION ) // (mm) Distance from real junction edge
  #else
    #define JUNCTION_DEVIATION_MM 0.017       // (mm) Distance from real junction edge (if you need to manually set this value)
  #endif
  //#define JD_HANDLE_SMALL_SEGMENTS          // Use curvature estimation instead of just the junction angle
                                              // for small segments (< 1mm) with large junction angles (> 135°).
#endif`

This would place the formula right inside the firmware at compile time and allow it to use the values set elsewhere (if they are set) instead of peeps trying to calculate them.  

One thing that's been on my mind about this topic is - If the JD value is based on ACCEL, shouldn't marlin should be calculating it based on the values set at the time of execution? Meaning rather than us setting JUNCTION_DEVIATION_MM by hand after doing human maths and test shapes, shouldn't we be able to set just a value (in this case classic jerk equiv) in the config and then marlin does the rest based on the results of M204 if stored in EEPROM and if not use the value of M201 and if not then use the value defined at compile with DEFAULT_ACCELLERATION? I bring this up because if, at compile time, my JD is 0.017 for an ACCEL value of 575 and a "classic jerk" value of 5.0, and then in my slicer I go and change the ACCEL to 1000.. have I not just skewed the formula that determined the 0.017 value to begin with? The new value of that combination would be 0.010 and not 0.017. This is the reason I never set it in cura.. 

Some sanity checks would need to be added as well .. but we can't have JD w/o ACCEL. 

As a final afterthought, if cura had a calculated field for the junction deviation setting based on enabling accel/jerk that would be epic.. but that's not for this github :-)