MarlinFirmware/Marlin #11606: [FR] Doubleclick As Move-Z and Babystep-Z

sarf2k4, 2018-08-22:
Hi,

I would like to add/enhance the idea of doubleclick feature that marlin currently has. 

Problems:
As of now, the doubleclick are default for babystep-z adjustment only and when we are not printing, we are still able to access this babystep adjustment. 
When not in printing, often we increase z-height for maintenance, by manually going into menu>prepare>move axis>move z> x mm, it is quite a lengthy process. I often have to stop printing midway because mishaps and after it has finished printing the print head just stopped there. After that I have to navigate through the menu each time I wanted to take off the failed parts. This also applies if we want to maintain other stuff that requires z axis movements. 

Proposal:
During printing, the doubleclick feature will become babystep-z adjustment.
When not in printing mode either by SD or USB such as octoprint or repetier, it will become move-z axis by 1mm per encoder step.

I am sure that this can be implemented since the "prepare" menu will turn into "tune" menu during movement and/or printing.

dot-bob, 2018-08-23:
The pending pull request #11191 implements this feature.

sarf2k4, 2018-08-23:
Is this feature included for 1.1.x?

AnHardt, 2018-08-23:
No. And it's not very likely this will happen, because all \*1.?.? branches are closed for new features.
Even #11191 (for bugfix-2.0.x) is not merged in yet.

sarf2k4, 2018-08-23:
Does that mean 1.x.x going to reach at the end of its development and moving to 2.x.x?

AnHardt, 2018-08-23:
**Yes**!

thinkyhead, 2018-08-27:
Lately I'm thinking, when printing double-click should go to babystep. When not printing it should go to the Move Axis menu. Optional, of course.

Roxy-3D, 2018-08-27:
Yeah....   I think this makes sense.    And it would be nice to have the Z-Axis part of the move menu be the selected option...    So...  In that case, it takes 3 clicks, but you are still adjusting the Z-Axis if you don't over ride things.

rajnagar92, 2018-08-28:
Going to the move axis menu seems to eliminate the simplicity of this feature. It's more intuitive to me to have it jog the Z-Axis in the same method used to Z-babystep. This would be useful to me nearly every time I use my printer.

Currently in Marlin, it seems like there are a lot of steps to do something as common as move the Z-axis when changing filament / clearing a print / cleaning the bed.

thinkyhead, 2018-08-30:
Reorganized menu in 2.0 now…