MarlinFirmware/Marlin #11609: SD card

Galax97, 2018-08-22:
Hello. 
I have a new Einsy RAMBo 1.0a motherboard and I configure Marlin v1.1.9 firmware on this board. I have a full graphic smart controller LCD. After changing configuration.h, I managed to set up the LCD but could not read the SD card. When I insert the SD card into the slot, the LCD writes "The card inserted" but I see the "No SD card" in the menu and I do not see the files on the card. The SD card is OK. My computer see the SD card and the files on the card. I've formatted the card on FAT32.
Can you help me please?

thetechnick, 2018-08-22:
I have the same issue with Marlin v1.1.9.
Serial Console prints: `echo:SD init fail` once when I insert the card after startup and twice when booting up with a inserted SD card.

My setup:
- MKS Gen v1.4
- RepRapDiscount Smart Controller
- TMC2130 Driver with SPI
- Anycubic Kossel Linear Plus (Delta)

There seems to be no Hardware defect on the LCD-Smart Controller because I already switched it with one I use in my Prusa i3, where the same SD cards do not have issues.

My Marlin config can be found here: https://github.com/thetechnick/Marlin/tree/anycubic-kossel-mksgen-tmc2130

thetechnick, 2018-08-22:
@Galax97 I just found my issue by reading this:
https://github.com/MarlinFirmware/Marlin/issues/11312#issuecomment-408145106

In my case using the TMC2130 drivers with SPI enabled will use the same pins (`53` and `49`) as the SD card communication. When using the `BOARD_MKS_GEN_L` config instead of `BOARD_MKS_GEN_13` the pins `59` and `63` will be used enabling the SD card reader to work on `53` and `49`.

There is also a comment in the file indicating a issue with the PIN selection:
```
// CS Pins wired to avoid conflict with the LCD
```

MrAlvin, 2018-09-02:
I am developing a new RAMPS 1.7 board, and I am also having issues with reading the SD card. For my testing, I am currentlyusing the  4x20 text LCD, ReprapDiscount Smart Controller LCD + SD-card system.  In Marlin configuration.h this is known as the REPRAP_DISCOUNT_SMART_CONTROLLER

I have made a small Arduino sketch, that directly adresses the SD-card, and does a "list filenames" from the card. 
Using this test sketch I can read the SD-card on my setup, so this tells me that the hardware works. 
The test sketch can be found here: https://github.com/MrAlvin/RAMPS_1.7/tree/master/Arduino%20test%20sketches/test%20ports-3-4-LCD/SD_listfiles

But I can not get Marlin to read the SD-card, so I will continue my research into why that might be.

MrAlvin, 2018-09-02:
hmmm....  after making very sure that the SDcard was placed just right in the card reader (I checked it was the case by using the SD_listfiles test sketch), then Marlin 1.1.9 was able to read the file list, and Repetier Host 2.0.5 reports all files with last name .gco

MrAlvin, 2018-09-02:
And while avoiding moving the SD-card, then I can also get bugfix-2.0 to read the SD-card. 
I am happy :-) 

Do you (@thetechnick or @Galax97 ) still have problems about reading your SD-cards?

thinkyhead, 2018-09-06:
Some people have worked around sluggish SD card readers by putting this at the end of the `setup()` function:
```cpp
#if ENABLED(SDSUPPORT)
  if (!card.cardOK) card.initsd();
#endif
```

boelle, 2019-02-20:
@Galax97 problem solved?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-03-17 by p3p