MarlinFirmware/Marlin #11611: Add support for FYSETC F6 V1.3 Board

Ccleorina87, 2018-08-22:
Add board for FYSETC F6 V1.3 Board

This link >> https://fysetc.github.io/F6_V1.3/

Above link have the info of the board, pins layout and ect.

I trying out this board but cant make it working on Bugfix-2.0.x and 1.1.9

closed 2018-08-23 by Ccleorina87

Ccleorina87, 2018-08-23:
I already get these board working so i close this issue...

thinkyhead, 2018-08-25:
That's a neat looking board. Is it basically a RAMPS variant? It says it works "out of the box" so I assume so.

Ccleorina87, 2018-08-27:
Im not good this programming but can understand basic one... the original firmware provided by the board is Marlin 1.1.8

thinkyhead, 2018-08-30:
@teemuatlut — Has the constructor signature changed for `TMC2208Stepper`?

This is what the Marlin code has now:
```cpp
  #define _TMC2208_DEFINE_SOFTWARE(ST) \
    TMC2208Stepper stepper##ST(ST##_SERIAL_RX_PIN, ST##_SERIAL_TX_PIN, ST##_SERIAL_RX_PIN > -1)
```

Ccleorina87, 2018-08-31:
i found out if i uncomment these it was able to compile. i might testing it today or day after day after tomorrow...

  #define X_HARDWARE_SERIAL  Serial1
  //#define X2_HARDWARE_SERIAL Serial1
  #define Y_HARDWARE_SERIAL  Serial1
  //#define Y2_HARDWARE_SERIAL Serial1
  #define Z_HARDWARE_SERIAL  Serial1
  //#define Z2_HARDWARE_SERIAL Serial1
  #define E0_HARDWARE_SERIAL Serial1
  #define E1_HARDWARE_SERIAL Serial1
  #define E2_HARDWARE_SERIAL Serial1
  //#define E3_HARDWARE_SERIAL Serial1
  //#define E4_HARDWARE_SERIAL Serial1

teemuatlut, 2018-08-31:
> Has the constructor signature changed for TMC2208Stepper?

No changes there.
Besides, Marlin is pulling an old v0.1.1 version anyway. Arduino IDE builds just fine.
Update the pulled version to `0.2.2`.

thinkyhead, 2018-08-31:
TMC2208Stepper 0.2.2 is causing compile errors in Travis CI. Can you help me figure out how to fix them? It affects the DUE, STM32F1, and Teensy35 builds.

https://travis-ci.org/MarlinFirmware/Marlin/jobs/423215870

Ccleorina87, 2018-09-01:
It now working well on new update 4 hour ago... thanks...

Suggy50, 2018-09-22:
Not any good with this sort off stuff so please forgive me if im wrong but you have the same pin asigned to 2 functions

#ifndef FILWIDTH_PIN
  #define FILWIDTH_PIN 10 // Analog Input on endstop X+
#endif

// Define digital pin 10 for the filament runout sensor. Use the digital input 10 on the servos connector
#define FIL_RUNOUT_PIN 10

Suggy50, 2018-09-22:
As it been imported yet?

AnHardt, 2018-09-22:
digital != analog

p3p, 2018-09-22:
Not all people understand AnHardt's syntax so to clarify, although the pins share the same identifier, digital and analogue pins are not the same.

Ccleorina87, 2018-09-24:
> As it been imported yet?

Imported what exactly??

Suggy50, 2018-09-24:
Fully working in Marlin 2
I have downloaded the latest bug fix and the FYSETC board ain't included in the pin's folder

Ccleorina87, 2018-09-24:
> Fully working in Marlin 2
> I have downloaded the latest bug fix and the FYSETC board ain't included in the pin's folder

It's fully working on both Marlin 1.1.9 and Marlin 2.... I'm using it on my D-Bot Printer... 
You need add manually and that's all i can say now...

Ccleorina87, 2018-09-24:
> Not any good with this sort off stuff so please forgive me if im wrong but you have the same pin asigned to 2 functions
>
> #ifndef FILWIDTH_PIN
> #define FILWIDTH_PIN 10 // Analog Input on endstop X+ [...]

About these already change to 
#ifndef FILWIDTH_PIN
  #define FILWIDTH_PIN 9 // Analog Input on endstop X+
#endif

mroote, 2018-09-24:
I've run into a few issues with this board, my X and Z sockets don't seem to work but I was able to work around that by changing the E1 and E2 sockets to X and Z.

Another issue I haven't figured out yet is I can't seem to turn on the cooling fans, I have one plugged into Fan0 port but running `M106` doesn't seem to turn it on. Anyone else having issues with the cooling fans?

thinkyhead, 2018-09-24:
If anyone has a confirmed-working pins file for this model, ZIP it up and send it our way and we'll add it to the project.

mroote, 2018-09-24:
I've opened a ticket with the vendor so hopefully they can provide the correct pins. They have a repo [here](https://github.com/FYSETC/Marlin1.1.9ForFYSETC_F6_TMC2130/blob/master/Marlin/pins_FYSETC_F6_13.h) but it has some differences from their documentation pinout [here](https://wiki.fysetc.com/F6_V1.3/#pin-definition). Hopefully they can clear up what pins are correct. I've tried both for the fan settings but neither seem to work for me.

Ccleorina87, 2018-09-25:
The documentation pin out on [Here](https://github.com/FYSETC/Marlin1.1.9ForFYSETC_F6_TMC2130/blob/master/Marlin/pins_FYSETC_F6_13.h) is correct... it was update 18 days ago to work on marlin 1.1.9

mroote, 2018-09-25:
That pinout didn't work for me, the `FAN0_PIN` variable should be `FAN_PIN` and the pins were 44, 45, 46 on my board.

mroote, 2018-09-25:
I've made a PR with my working pinout https://github.com/MarlinFirmware/Marlin/pull/11920 if someone could also test it that would be great!

Ccleorina87, 2018-09-25:
> I've run into a few issues with this board, my X and Z sockets don't seem to work but I was able to work around that by changing the E1 and E2 sockets to X and Z.
> 

Meaning that X and Z pin swap with E1 and E2 fix it?

mroote, 2018-09-25:
Yes that fixed it for me, but I think that's an issue with those sockets on my board as I haven't seen other people with this issue. I'm talking to the vendor now to see if they know anything or can send a replacement.

Ccleorina87, 2018-09-25:
your board might have different revision or faulty board... because mine work find with original X and Z pin...

mroote, 2018-09-25:
Yeah I suspect it's faulty as swapping the ports worked fine so it's not the drivers. I also checked all the jumpers and they are in the right places too.

Ccleorina87, 2018-09-28:
> Yeah I suspect it's faulty as swapping the ports worked fine so it's not the drivers. I also checked all the jumpers and they are in the right places too.

So did the seller send new board?

daboi, 2018-10-12:
does anybody tried running klipper on f6 board?

thinkyhead, 2018-10-12:
There are better venues to discuss Klipper. This issue queue is not the place.

daboi, 2018-10-12:
@thinkyhead sorry about that. I just couldn’t find any info about this board online and wanted to ask users here. already deleted the comment.

skaaj4, 2018-11-22:
> I've run into a few issues with this board, my X and Z sockets don't seem to work but I was able to work around that by changing the E1 and E2 sockets to X and Z.

Hello, i have the same issue as Mr. mroote.
I can use my tmc2130 drivers with Y, E0, E1 and E2. The X- and Y-axis do just work with a standard driver (A4899). Swapping the pins worked for me too, but this not really a solution for me because i also use tmc2130 drivers for my two extruders. Does anybody has new informations about this board?

I used the current bugfix-2.0.x version.

Kadah, 2018-11-22:
I've got an F6 on order. My printer will need all the steppers, so I'll figure it out when I get it if its not already by then.

ErikFontanel, 2018-11-24:
To add to @skaaj4, if you prefer platform.io + VSCode or Atom instead of the Arduino IDE I've managed to compile both Marlin 1.1.9 and bugfix-2.0.x branches for the Fysetc F6 by converting the FYSETC F6 Board plugin to a platform.io configuration.  
[Read the instructions here](https://gist.github.com/ErikFontanel/a81807afee6f6bc8db15f53d1577a06f)

skaaj4, 2018-11-24:
Thank you @ErikFontanel, i already figured it out how to fix that. I created a pull request for platformio, so it should be work out of the box on the next pio update.

Kadah, 2018-12-12:
Setting up PIO since I'm running in to the path length bug with ArduinoIDE and old workaround for that causes some other issues.

Till that PIO update, I used ErikFontanel's workaround, but with a couple corrections to make it work. 'fysetcf6_13.json' needed to be renamed 'fysetc_f6_13.json' to match the env string Marlin2 uses. With a fresh install of PlatformIO Core, needed to run a build on 'megaatmega2560' first have PIO download the needed 'framework-arduinoavr' files, then create 'variants/fysetcf6/pins_arduino.h'.

florin03, 2019-07-18:
Hello,
Any one managed to define pins for fans? for example part cooling is fan0, and changing pin-out (to use heater 1 pin......) in pin file do nothing.
Thank you